eclipse-threadx/threadx #114: Please add the CMake files for cortex_a5

pandersb, 2021-06-08:
Hi,

I managed to build and run ThreadX on the Atmel SAMA5D3 Xplained, but it was unnecessarily hard work.
It would be wonderful if you could provide the CMake files for all the supported ports.

I had to manually create cmake/cortex_a5.cmake and ports/cortex_a5/gnu/CMakeLists.txt.
While this wasn't complicated it involves setting up a local git branch which must be maintained and merged with master on every update.

hwmaier, 2021-06-10:
@pandersb FYI, I made a similar request for cortex_m33 here https://github.com/azure-rtos/netxduo/issues/35

goldscott, 2021-06-10:
Hello, sorry to tell you that currently we don't plan to maintain CMake for ports other than cortex m0/3/4/7. This may change in the future.

closed 2021-06-10 by goldscott