eclipse-threadx/threadx #116: Threadx dynamic application loading cannot pre-initialize default values

GetFreedomPro, 2021-06-09:
The Applications code uses the structure to initialize the default values. The application will not run
How to solve this problem？

example：
`typedef struct test_11{
    ULONG   test1;
    ULONG   test2;
    ULONG   test3;
    ULONG   test4;
}test_22;

test_22 test444 = {
    123,
    456,
    789,
    147,
};`

goldscott, 2021-06-09:
What processor & toolchain are you using?

GetFreedomPro, 2021-06-10:
Processor: STM32H743 (M7)
Tool chain: arm-none-eabi @goldscott

goldscott, 2021-06-25:
Is the `_gcc_setup` function running correctly? Do you see it copying this data from code to data memory?

GetFreedomPro, 2021-07-01:
The reference example:
	threadx/ports_module/cortex_m7/gnu/example_build/

File path:
	threadx/ports_module/cortex_m7/gnu/example_build/sample_threadx_module.c

	/* Define the counters used in the demo application...  */

	ULONG           thread_0_counter = 0xAAAAAAAA;
	ULONG           thread_1_counter = 0xCCCCCCCC;
	ULONG           thread_1_messages_sent;
	ULONG           thread_2_counter = 0x55555555;
	ULONG           thread_2_messages_received;
	ULONG           thread_3_counter;
	ULONG           thread_4_counter;
	ULONG           thread_5_counter;
	ULONG           thread_6_counter;
	ULONG           thread_7_counter;
	ULONG           semaphore_0_puts;
	ULONG           event_0_sets;
	ULONG           queue_0_sends;

File path:
	threadx/ports_module/cortex_m7/gnu/example_build/sample_threadx_module_manager.c

	/* Define fault handler.  */

	VOID module_fault_handler(TX_THREAD *thread, TXM_MODULE_INSTANCE *module)
	{

		/* Just increment the fault counter.   */
		memory_faults++;
	}
	
Test result:
	
1.	
	ULONG           thread_0_counter;
	ULONG           thread_1_counter;
	ULONG           thread_1_messages_sent;
	ULONG           thread_2_counter;
	
	the application module runs normally
	
--------------------------------------------------------	
2.	
	ULONG           thread_0_counter = 0xAAAAAAAA;
	ULONG           thread_1_counter = 0xCCCCCCCC;
	ULONG           thread_1_messages_sent;
	ULONG           thread_2_counter = 0x55555555;
	
	Do you see it copying this data from code to data memory?
	
	Yes, in memory and I can see 0xAAAAAAAA, 0xCCCCCCCC and 0x55555555
	
	But the module_fault_handlers function is still entered, making the thread unable to run

GetFreedomPro, 2021-07-05:
Anything else to add?
I'm very interested in the dynamic application of Threadx

goldscott, 2021-07-14:
Hi @GetFreedomPro - sorry for the delay. Where exactly is the memory fault occurring? You can put a breakpoint in MemManage_Handler in tx_thread_schedule.S. This handler will call _txm_module_manager_memory_fault_handler. Can you also give us the contents of _txm_module_manager_memory_fault_info.

GetFreedomPro, 2021-07-15:
Hi，@goldscott
+-----------------------------------------------------------------------------+
|                                                                                                                   |
|       tx_thread_schedule.S                                                                           |
|                +                                                                                                 |
|                |                                                                                                   |
|                v                                                                                                  |
|       UsageFault_Handler:                                                                            |
|                +                                                                                                 |
|                |                                                                                                   |
|                |                                                                                                   |
|                +-> LDR     r12,=_txm_module_manager_memory_fault_info    |
|                |                                                                                                   |
|                |                                                                                                   |
|                +-> BL      _txm_module_manager_memory_fault_handler        |
|                                                                                                                    |
|                                                                                                                    |
+------------------------------------------------------------------------------+

_txm_module_manager_memory_fault_info :

	txm_module_manager_memory_fault_info_thread_ptr		      TX_THREAD *	0x24001d28 (Hex)	
	txm_module_manager_memory_fault_info_code_location	      void *	0x2400673c (Hex)	
	txm_module_manager_memory_fault_info_shcsr			      ULONG	0x70001 (Hex)	
	txm_module_manager_memory_fault_info_cfsr			       ULONG	0x82 (Hex)	
	txm_module_manager_memory_fault_info_mmfar			       ULONG	0x53f06080 (Hex)	
	txm_module_manager_memory_fault_info_bfar			       ULONG	0x53f06080 (Hex)	
	txm_module_manager_memory_fault_info_control		        ULONG	0x1 (Hex)	
	txm_module_manager_memory_fault_info_sp				ULONG	0x24008a38 (Hex)	
	txm_module_manager_memory_fault_info_r0				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r1				ULONG	0x24008a98 (Hex)	
	txm_module_manager_memory_fault_info_r2				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r3				ULONG	0x53f06080 (Hex)	
	txm_module_manager_memory_fault_info_r4				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r5				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r6				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r7				ULONG	0x24008a58 (Hex)	
	txm_module_manager_memory_fault_info_r8				ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r9				ULONG	0x24008800 (Hex)	
	txm_module_manager_memory_fault_info_r10			       ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r11			       ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_r12			       ULONG	0x0 (Hex)	
	txm_module_manager_memory_fault_info_lr				       ULONG	0x24006db3 (Hex)	
	txm_module_manager_memory_fault_info_xpsr			       ULONG	0x61000000 (Hex)

goldscott, 2021-07-15:
Where is the memory fault happening? in tx_thread_schedule? looks like it is barfing on data access at address 0x53f06800.

GetFreedomPro, 2021-07-16:
Yes, 0x53F06080 is an illegal address
If you import a dynamic application using the TXM_MODULE_MANAGER_MEMORY_LOAD () function, it is not supported to define the default value of a global variable in a module. This function only copies the code area, not the data area

goldscott, 2021-07-20:
Hi @GetFreedomPro - I see that the gcc_setup function is not position-independent. I'm not sure how it works even when there is no initialized data. We're discussing and fixing this internally.

goldscott, 2021-07-22:
We have fixed this and it will be available in our next release in early August.

@GetFreedomPro - try doing these two things: build with -mno-pic-data-is-text-relative (rather than -mpic-data-is-text-relative), and in the linker file, make sure the RAM segment is higher than FLASH segment, e.g.:
```
  __FLASH_segment_start__ = 0x00030000;
  __FLASH_segment_end__   = 0x00040000;
  __RAM_segment_start__   = 0x10000000;
  __RAM_segment_end__     = 0x10008000;
```
Let me know if that works.

closed 2021-07-22 by goldscott