eclipse-threadx/threadx #119: Writing queue from interrupted causes hard fault

atul6019, 2021-06-15:


atul6019, 2021-06-15:
Hi, i am using cortex m7 board facing issue while writing from writing from isr.

goldscott, 2021-06-15:
Please give us more details. Have you done any debugging/stepping through the code to find exactly where this is happening?

atul6019, 2021-06-16:
Hi , thanks for your response. In my case i have common task handler  on which i am waiting for queue data. if data is available i am  calling actual task.  when i am writing queue from ISR it goes to hardfault from   _tx_thread_system_return(); API. as combined_flags is 0.

`                                TX_THREAD_SYSTEM_RETURN_CHECK(combined_flags)
                                if (combined_flags == ((ULONG) 0))
                                {

#ifdef TX_THREAD_ENABLE_PERFORMANCE_INFO

                                    /* There is another thread ready to run and will be scheduled upon return.  */
                                    _tx_thread_performance_non_idle_return_count++;
#endif

                                    /* Preemption is needed - return to the system!  */
                                    _tx_thread_system_return();
                                }
`

goldscott, 2021-06-25:
what does your call to tx_queue_send look like in your ISR? are you calling any other ThreadX functions in your ISR?

goldscott, 2021-07-29:
No response, closing.

closed 2021-07-29 by goldscott