eclipse-threadx/threadx #120: problem with comments of ports/cortex_m7/ac5/src/***.s

yanglx2022, 2021-06-25:
for example, `;` is not used for comments of [tx_thread_context_restore.s](https://github.com/azure-rtos/threadx/blob/master/ports/cortex_m7/ac5/src/tx_thread_context_restore.s)
f5056f4923410b99c87ab909d24ae77bf41923b6

build output (uVision v5.30.0.0 ac5)

> \ports\cortex_m7\ac5\src\tx_thread_context_restore.s(1): error: A1167E: Invalid line start 

> \ports\cortex_m7\ac5\src\tx_thread_context_restore.s(3): error: A1163E: Unknown opcode Copyright , expecting opcode or Macro

goldscott, 2021-06-25:
AC5 assembly code now uses the C preprocessor. On the command line/project options, the following is used to tell the tools to pass the assembly through the C preprocessor and if you need to define symbols, this is how:

`--cpreproc --cpreproc_opts=-D,TX_EXECUTION_PROFILE_ENABLE --cpreproc_opts=-D,TX_ENABLE_EXECUTION_CHANGE_NOTIFY`

closed 2021-06-25 by goldscott

yanglx2022, 2021-06-25:
thanks a lot, it works with option `--cpreproc`