eclipse-threadx/threadx #121: Mismatching version number between CMakeLists.txt and tx_api.h

GPRESean, 2021-06-27:
CMakeLists.txt file contains the version of the project as 6.0.0
https://github.com/azure-rtos/threadx/blob/244365fc6a937c754cfc6bb8e6895dc6f37bac0e/CMakeLists.txt#L3-L7

Which is differ from the version from tx_api.h
https://github.com/azure-rtos/threadx/blob/244365fc6a937c754cfc6bb8e6895dc6f37bac0e/common/inc/tx_api.h#L104-L107

It seems to me that version description is duplicated and placed in two placed. Additionally, the CMake project version is outdated and not being used.

In my humble opinion, it would be less confusing to do one of the following
- Idea1: Update the CMake project version and replace literal version numbers in tx_api.h with CMake project version. e.g. [Adding a Version Number and Configured Header File](https://cmake.org/cmake/help/latest/guide/tutorial/A%20Basic%20Starting%20Point.html#adding-a-version-number-and-configured-header-file)
- Idea2: Remove CMake project version in CMakeLists.txt and maintain version in tx_api.h

GPRESean, 2021-06-27:
Oh the version number is also in here
https://github.com/azure-rtos/threadx/blob/244365fc6a937c754cfc6bb8e6895dc6f37bac0e/common_smp/inc/tx_api.h#L109-L112

goldscott, 2021-07-27:
We're going to remove version from cmake.

closed 2021-07-27 by goldscott

goldscott, 2021-08-04:
We have made this change internally and it will be available in the next ThreadX release, sometime in early October.