eclipse-threadx/threadx #122: Why queue message size is restricted to maximum of 16 32-bit word?

RoshDevi, 2021-06-29:
I use the API "tx_queue_create" to create the queue and maximum message size of the queue is up to 16 32-bit word size.
Is there any reason why queue message size is restricted to 1 - 16 32-bit word?
In other RTOS, we see that there is no restriction on the message size.

goldscott, 2021-06-29:
See here for more info: https://docs.microsoft.com/en-us/answers/questions/292596/about-the-limite-of-the-message-size-of-queue.html

RoshDevi, 2021-06-30:
@goldscott - Thanks for the information.
I have defined macro - TX_DISABLE_ERROR_CHECKING, which is calling "_tx_queue_create" API instead of "_txe_queue_create". I see that error checks are disabled in the "_tx_queue_create". If I try to pass a message size of 1024 32-bit words ( 4096 bytes ), it is getting created successfully. And I can do send & receive operation. 
Apart from the error check, what is the change in "_txe_queue_create"? Is it safe to trust it without error check that queue operations always succeed? Or Any limitations in the usage of "_tx_queue_create" ?

RoshDevi, 2021-06-30:
Also, in documentation, it is said that message size can be 1 - 16 32-bit words. But with the error check disabled, we can create a message size of any 32-bit words.

hwmaier, 2021-06-30:
My guess is that the size limit is there to keep the latency down as larger messages would block the kernel for a longer period of time during the copy operation.
If the latency is not of concern to a user, one could replace TX_16_ULONG with a TX_MAX_MSG_SIZE macro defaulting to 16 but which can be redefined by the user. That would allow a user to override the size in tx_user.h.

RoshDevi, 2021-06-30:
@hwmaier  - Thanks for the update. 
One more query I have is though we have the provision of increasing the message size. Still, there is a restriction that message size should be multiple of 4 bytes ( 32 bit word ). 
Lets say I want by queue message size to be 65 bytes. Can I do this? Or should I make it to 68 bytes ( 17 32-bit words )?

RoshDevi, 2021-07-01:
> @goldscott - Thanks for the information.
> I have defined macro - TX_DISABLE_ERROR_CHECKING, which is calling "_tx_queue_create" API instead of "_txe_queue_create". I see that error checks are [...]

Hello, It is blocking our development. Can you please answer this question asap ? 
Also, we want to enable error checking along with an increase in message size of the queue, is there any other way to achieve it other than changing TX_16_ULONG?

goldscott, 2021-07-01:
All of the txe functions are error-checking shells. You can read more about that here: https://docs.microsoft.com/en-us/azure/rtos/threadx/chapter2#configuration-options

Your queue size should be a multiple of 4 bytes. If your message size is 65 bytes, make the queue message size 68 bytes.

RoshDevi, 2021-07-02:
Thanks @goldscott.
We have one more query - We want to enable the error checking along with an increase in the size of the message queue. One way we can see is by modifying the macro TX_16_ULONG locally. Also, this macro is currently used for only queue message size checks. 
If we change this macro value, can you tell if any implications will be seen in other features?

goldscott, 2021-07-02:
Hi @RoshDevi 
You can do one of two things: (1) redefine TX_16_ULONG to whatever size you need, or (2) edit the txe_queue_create function and replace TX_16_ULONG with your own definition or magic number. 
You'll have to remember to do this again if you ever update the ThreadX source code. I suggest option 2 because we could use TX_16_ULONG for something else in the future (though this isn't likely).

RoshDevi, 2021-07-06:
Thanks @goldscott. It is clear now. 
One suggestion from our side, try to make the upper limit check on queue message size as configurable macro which can be overridden by tx_user.h file.

closed 2021-07-06 by RoshDevi