eclipse-threadx/threadx #123: build Cortex M3 example with GHS

chenxy1988, 2021-06-30:
Hi Team,

Recently, I dowonloaded latest version threadx for github, and built it with my GHS MULTI environment. 
I am using example project in ports\cortex_m3\ghs\example_build\azure_rtos_workspace.gpj

But below error message happens,

"
"..\..\..\..\common\src\tx_thread_create.c", line 276: error #20: identifier
          "__IPSR" is undefined
          if (TX_THREAD_GET_SYSTEM_STATE() >= TX_INITIALIZE_IN_PROGRESS)
              ^
"

Could you please help to check whether I need do some configuration in my Multi workspace or not?

Many thanks!

Br,
Xiangyu

chenxy1988, 2021-06-30:
Forgot to tell you my Multi version is 5.06, is that too old to use latest threadx GHS sample?

andrejm-rtos, 2021-07-26:
I do not have access to MULTI 5.06. We are using MULTI 7. I just tried with MULTI 7.1.6 and the latest version builds correctly.