eclipse-threadx/threadx #130: Calling tx_semaphore_create again after tx_semaphore_delete causes hard fault

yanglx2022, 2021-07-28:
[image]

goldscott, 2021-07-28:
Please provide more info. Where is crash occurring? Did the delete call return successfully? Where are you calling delete and create?

yanglx2022, 2021-07-29:
The calling process is as follows:
1. call `tx_semaphore_create()` before `tx_kernel_enter()`, it returns `TX_SUCCESS`
2. call `tx_semaphore_delete()` in a thread, it returns `TX_SUCCESS`
3. call `tx_semaphore_create()` in the same thread, crash happens

Please see the screenshot above where the crash occurred. The direct reason is that the pointer `next_semaphore = 0` was accessed.

goldscott, 2021-07-29:
Oh, well, there's your problem - do NOT call any ThreadX APIs before calling tx_kernel_enter.

closed 2021-07-29 by goldscott