eclipse-threadx/threadx #133: Add .cpu <arch> directive to .S files

xiongyu0523, 2021-08-03:
When I'm porting threadX to Arduino environment, some of the core platform's toolchain does not specify -mcpu option for assembly code so it cause gcc throw an error related to 'MSR' and 'MSR' instructions. I already submit a PR to a known core platform to add -mcpu but there are too many core platform Arudino support and old version in use, so better if we can add this directive to all ports (I assume this is not harmful) so it will be fixed even -mpcu is not added

Related PR:
https://github.com/arduino/ArduinoCore-samd/pull/642

goldscott, 2021-08-03:
Internally, we have one copy of the file (e.g. for ARMv7-M) that is used for M0, M3, M4, M7. So we cannot put the cpu in the file. Is there a different way to do this? We could put the arch in the files.... would that help?

xiongyu0523, 2021-08-04:
Are you saying you are generating Cortex-Mx CPU ports from a single ARMv7-M arch base file? I was thinking there're different source and implementation as cortex-m0 only implements a small number of Thumb-2.

What is the directive for specify arch? I can have a quick try if it can suppress GNU AS error

goldscott, 2021-08-04:
Sorry, that should be M3/4/7. Not M0.

Check out the -march option here: https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html

xiongyu0523, 2021-08-05:
this option is for compiler, what I was mentioning is Arduino has many core platform (with toolchain and its configurations), it is not  easy to revise all of them support -mcpu or -march. If we can add  .cpu cortex-m0 to our port source, it can fix this without need specifying them.

xiongyu0523, 2021-08-05:
Ok. I found there are equivalent method, and source code directive take priority over compiler options.
I have tried on one of the platform generating errors, by adding `.arch armv6-m` on top of .S source, it fixes the error.

goldscott, 2021-08-05:
I don't know what the arduino build environment looks like. Couldn't you pass a compiler flag to that to specify the cpu/architecture?
Or you can edit the ThreadX source and create a pull request, we will review and possibly implement it.

xiongyu0523, 2021-08-20:
I add this directive out side of official port and then include official port to solve this problem. we can close, thanks

closed 2021-08-20 by xiongyu0523