eclipse-threadx/threadx #134: Error occurs after ThreadX enables TraceX

arilink, 2021-08-12:
Development environment:
windows10 x64 mdk v5.34
threadx usbx 6.16
My project has entered the later stage of development and I want to try to use TraceX to optimize the program. The current project uses ThreadX+USBX. When I try to open #define TX_ENABLE_EVENT_TRACE in the tx_user.h file and compile, the following error appears. I did not find a related solution. Method May I ask what caused such a problem?

arilink, 2021-08-12:
[image]

goldscott, 2021-08-12:
Hello @arilink - have you #defined UX_TRACE_INSERT_MACROS?

arilink, 2021-08-13:
Is it not enough to just #define TX_ENABLE_EVENT_TRACE? I only found the use of the software in the TraceX documentation, but there is no operating method in the program

arilink, 2021-08-13:
My operating system is windows. Do I need #defined UX_TRACE_INSERT_MACROS?

xiaocq2001, 2021-08-13:
In USBX, trace information is generated via USBX trace functions that are enabled by UX_TRACE_INSERT_MACROS

arilink, 2021-08-13:
Ok, I didn't find the relevant information in the document. What if it is FileX NetX?

TiejunMS, 2021-08-13:
There is no additional macros for FileX or NetX.

arilink, 2021-08-13:
Understand, thank you very much for your help

closed 2021-08-13 by arilink