eclipse-threadx/threadx #136: add API to check whether a kernel object is valid or not valid

xiongyu0523, 2021-09-07:
Scenario: 

When I'm porting threadx to some project using freertos or cmsis-os, a common use case is code need to check whether an kernel object is valid or not valid. 

Problem:

For each service in threadx, there is a id can represent the validation of that object, (i.e. tx_semaphore_id for semaphore) but the macro value is defined privately (i.e. TX_SEMAPHORE_ID), which make me hard to know if a static allocated object is deleted or not. Currently I use `!= TX_CLEAR_ID` but it is not accurate. 

Proposal:
Offer a series of service API to check this and return if a object is valid or not valid.

yuxin-azrtos, 2021-09-07:
ThreadX APIs validates the object before operating on it. Therefore we don't have APIs.  Suggest that developers check return values from ThreadX APIs and detect whether the object is valid or not.

closed 2021-09-09 by goldscott

xiongyu0523, 2021-09-14:
Base on this information, I can use xxx_info_get() function with all parameters set to NULL for object availability checking to satisfy middleware requirements.