eclipse-threadx/threadx #145: Storing registers of idle context

amgross, 2021-11-09:
Why are you storing registers of idle context? why not to store those registers just after ensuring it wasn't idle?
https://github.com/azure-rtos/threadx/blob/b216ceb25e2b6ae1ea83c1ae91d114a6fc2fc4ec/ports/arc_em/metaware/src/tx_thread_context_save.s#L153-L158

TiejunMS, 2023-03-14:
No activity for a long time. Closing.

amgross, 2023-03-14:
@TiejunMS why to close? Didn't got answer here yet...

TiejunMS, 2023-03-14:
Sorry about that. I'm closing issues those are not active for a long time. I will reopen this issue.

fdesbiens, 2026-09-08:
Apologies for the very long silence on this one, @amgross, and thanks for your patience.

The four stores in question are `r30`, `r25`, `r12` and `r11`, and they sit between `ld r1, [gp, _tx_thread_current_ptr@sda]` and the `breq r1, 0, __tx_thread_idle_system_save` that consumes `r1`. That placement is deliberate: it is instruction scheduling to hide the load-use latency of the pointer fetch. If the branch immediately followed the load, the pipeline would stall waiting for `r1`. Filling that window with stores that the interrupted-thread path has to perform anyway makes them effectively free on the path that matters.

You can see the same idiom a few instructions earlier in the same routine. `ld r0, [gp, _tx_thread_system_state@sda]` is followed by `st r3` and `st r2` before `breq r0, 0, __tx_thread_not_nested_save` consumes `r0`. The port is written throughout with explicit pipeline scheduling, which is also why you see the delay-slot forms `j.d`, `bl.d` and `j_s.d` everywhere rather than plain jumps. The identical sequence appears in `ports/arc_hs/metaware/src/tx_thread_context_save.s`, so it is a consistent choice across the ARC ports rather than an accident in one file.

The trade is asymmetric on purpose. The common, latency-sensitive case is an interrupt arriving while a thread is running, and that path needs those four registers saved regardless. The case you are asking about is an interrupt arriving in the scheduler idle loop, where the system had nothing to run anyway, so four redundant stores cost nothing that anyone can observe. Reordering to test first would remove those four stores from the idle path at the price of a pipeline stall on every interrupt taken in the hot path.

They are also harmless. The interrupt stack frame is already allocated before `_tx_thread_context_save` is entered, so the stores always land inside a valid 160-byte frame. On the idle path that frame is either discarded wholesale by `add sp, sp, 160` at `__tx_thread_not_nested`, or fully populated by `__tx_thread_nested_save` when the special nesting condition is detected. In the latter case those same four registers are simply written a second time with unchanged values.

I am closing this as answered, but please reopen or comment if you think the analysis is wrong, or if you have measurements showing the idle path matters in your application.

closed 2026-09-08 by fdesbiens