eclipse-threadx/threadx #148: add TXM_XXX_CALL_NOT_USED defines to module calls

amgross, 2021-11-14:
You added in the kernel side defines to uncompile system calls ( #67 ), can you please add those defines also in the module function? for example here:
https://github.com/azure-rtos/threadx/blob/b216ceb25e2b6ae1ea83c1ae91d114a6fc2fc4ec/common_modules/module_lib/src/txm_byte_pool_prioritize.c#L63

It is important to prevent runtime bugs where module compiles because in its compilation the system call exist, but the kernel don't has this system calls so it returns TX_NOT_AVAILABLE.

Always prefer compilation errors upon runtime error

goldscott, 2021-11-19:
Thank you for the suggestion. We are working on this and it will be available in our next release.

amgross, 2021-11-21:
thank you, it will make life easier

goldscott, 2022-07-15:
These TXM_*_CALL_NOT_USED defines were added to the module code in release 6.1.10. Closing.

closed 2022-07-15 by goldscott