eclipse-threadx/threadx #149: Stack of _tx_execution_isr_enter in arc port is not the system stack

amgross, 2021-11-25:
In version 6.1.9 you changed in the scheduler the call to _tx_execution_thread_enter will be on the system stack. same behavior that those functions runs on system stack is at context_restore function.

Is there is a good reason that in context save  _tx_execution_isr_enter  is not on system stack? (it may be problematic if got interrupt when the stack is already full)

TiejunMS, 2023-03-14:
No activity for a long time. Closing.

amgross, 2023-03-14:
@TiejunMS I still didn't got answer her. Why to close?

TiejunMS, 2023-03-14:
Sorry about that. I'm closing issues those are not active for a long time. I will reopen this issue.

fdesbiens, 2026-09-08:
Thanks for the report, and apologies for the very long delay in answering it.

You are right, and there is no good reason for the difference. `_tx_thread_schedule` was reordered in 6.1.9 so that `_tx_execution_thread_enter` runs on the system stack, but the equivalent reorder was never applied to `_tx_thread_context_save`. The other two callout sites in that same routine, the nested save and the idle system save, already run on the system stack, as does the `_tx_execution_isr_exit` call in `_tx_thread_context_restore`. Only the path where an interrupt preempts a running thread was left calling out on the thread's own stack.

Your concern about stack exhaustion is the right one. The callouts are supplied by the application, so ThreadX cannot bound their stack usage, and that usage was being charged to whichever thread happened to be running when the interrupt arrived, on top of the 160-byte interrupt frame the port had already allocated there.

I have opened #700 to move the switch to the system stack ahead of the callout in the ARCv2_EM, ARC_HS and SMP ARC_HS ports. `_tx_thread_context_fast_save` is unchanged, since the fast interrupt path never switches stacks by design.

closed 2026-09-08 by fdesbiens