eclipse-threadx/threadx #150: Allocating stack before calling function in arc_em assembly

amgross, 2021-11-25:
In multiple places I see in arc_em assembly you are allocating 16 bytes on the stack before calling c function, for example at:
https://github.com/azure-rtos/threadx/blob/b216ceb25e2b6ae1ea83c1ae91d114a6fc2fc4ec/ports/arc_em/metaware/src/tx_thread_schedule.s#L137-L139
https://github.com/azure-rtos/threadx/blob/b216ceb25e2b6ae1ea83c1ae91d114a6fc2fc4ec/ports/arc_em/metaware/src/tx_timer_interrupt.s#L195-L197
what is the reason for it?

TiejunMS, 2023-03-14:
No activity for a long time. Closing.

amgross, 2023-03-14:
@TiejunMS I still didn't got answer her. Why to close?

TiejunMS, 2023-03-14:
Sorry about that. I'm closing issues those are not active for a long time. I will reopen this issue.

fdesbiens, 2026-09-08:
Hi @amgross, and apologies for the very long delay in getting you a real answer.

Short version: those 16 bytes are not tied to argument passing — every one of those callouts is a `VOID f(VOID)`. They are a deliberate, port-wide safety convention inherited from the original Express Logic ARC port, and they are documented as a requirement in `ports/arc_em/metaware/readme_threadx.txt`, section 7 ("Interrupt Handling"): *"If it is written in C, you must allocate 16 bytes of stack space before it is called. This must also be recovered once your C ISR returns."*

The reason is that these assembly routines have no stack frame of their own, and at the point of the call `sp` is pointing directly at live data that the assembly code still needs after the call returns. In `_tx_thread_system_return` the solicited frame occupies `[sp, 0]` through `[sp, 72]` and has just been written; in `_tx_thread_schedule` `sp` is the system stack pointer; in an application ISR it is the 160-byte interrupt frame. Reserving four words before transferring control to compiler-generated code guarantees a margin between that live data and whatever the callee does with the stack, so the assembly can rely on its offsets being intact on return.

You can see the same convention in the frame layouts themselves: the ISR template stores `blink` at `[sp, 16]`, not `[sp, 0]`, and the nested path in `_tx_thread_context_save` does `sub sp, sp, 32` and then `st blink, [sp, 16]`. Offsets 0–15 are always left free for the callee. Sixteen bytes matches the size of the register-argument area for a reduced-register-set ARC build (r0–r3), which is where the number comes from.

To be completely transparent: the ARCv2 System V ABI does not itself require the *caller* to reserve this area — a conforming callee allocates its own register-parameter save area. So this is belt-and-braces rather than a strict ABI obligation. It costs two instructions, one of which sits in the branch delay slot (`bl.d` + `sub` in the slot), so it is essentially free on these paths, and it makes the assembly robust against toolchain or configuration differences.

The practical takeaway is the one in the readme: if you write an ARC ISR in assembly and call into C from it, do the same thing.

I am closing this as answered, but please reopen or comment if you would like to dig further into any of it.

closed 2026-09-08 by fdesbiens