eclipse-threadx/threadx #155: How to properly implement idle thread for resources cleanup?

MaJerle, 2021-12-23:
In the context of multiple network communication, several threads must be started and stopped, often same thread function is used with multiple instances in parallel.

Then dynamic allocation of thread handle and respective stack is a must, but not really supported by ThreadX.

There is thread entry/exit notification capability, however it cannot be used to cleanup thread stack pointer and thread instance as instance is used immediately, just after exit-callback function returns to the caller.

FreeRTOS has this solved more than just well, indeed it requires more memory resources. it implements idle thread, in charge of resource cleanup process for dynamically allocated threads.

I did a prototype of my implementation and would like to get a feedback from Microsoft, if this is the right usage, namely:

Code concept as below. Prototype seems to work properly.

```c
/* Main application byte pool */
#define TX_APP_MEM_POOL_SIZE                     (1024 * 32)
static UCHAR tx_byte_pool_buffer[TX_APP_MEM_POOL_SIZE];
static TX_BYTE_POOL tx_app_byte_pool = {0};

/* Idle thread uses pre-defined handle structure and statically defined stack memory */
static TX_THREAD idle_thread_handle;
static uint8_t idle_thread_stack[512];
static void idle_thread_func(ULONG param);

/* Thread app entry function */
static void app_thread_func(ULONG param);

/**
 * \brief           Idle thread function entry
 * \param[in]       param: User argument, don't care
 */
static void
idle_thread_func(ULONG param) {
    TX_INTERRUPT_SAVE_AREA
    TX_THREAD* t;

    while (1) {

        /* Process interrupts and check if anything to do */
        TX_DISABLE

        /*
         * Check a linked list of all created threads in the system.
         * Process threads in terminated or completed state.
         *
         * These threads use dynamic memory allocation,
         * hence it is now good time to clean them up
         */
beg:
        t = _tx_thread_created_ptr;
        while (t != NULL) {
            if (t->tx_thread_id != TX_THREAD_ID) {
                break;
            }
            if (t->tx_thread_state == TX_TERMINATED
                    || t->tx_thread_state == TX_COMPLETED) {
                TX_RESTORE

                /* Do we need a call to delete? */
                tx_thread_delete(t);

                /* Cleanup resources */
                /* How should we determine if allocation was dynamic or static? */
                tx_byte_release(t->tx_thread_stack_start);
                tx_byte_release(t);

                TX_DISABLE

                /* Start from beginning since linked list changed now after delete operation */
                goto beg;
            }
            /* Find next thread */
            if (t->tx_thread_created_next != t
                    && t->tx_thread_created_next != _tx_thread_created_ptr) {
                t = t->tx_thread_created_next;
            } else {
                t = NULL;
            }
        }
        TX_RESTORE

        /* Yield thread */
        tx_thread_relinquish();
    }
}

/**
 * \brief           Define the initial system.
 * \param           first_unused_memory: Pointer to the first unused memory
 */
VOID
tx_application_define(VOID* first_unused_memory) {
    TX_THREAD* t_handle;
    void* stack_mem;

    /* Initialize debug unit */
    debug_init();

    /*
     * Create idle thread to monitor all other threads, having lowest priority
     *
     * Idle thread has lowest priority and static handle/stack objects
     */
    tx_thread_create(&idle_thread_handle, "idle_thread", idle_thread_func, 0,
                     idle_thread_stack, sizeof(idle_thread_stack), TX_MAX_PRIORITIES - 1, 0, TX_NO_TIME_SLICE, TX_AUTO_START);

    /* Create byte pool for ThreadX and application handling */
    if (tx_byte_pool_create(&tx_app_byte_pool, "Tx_app_mem_pool", tx_byte_pool_buffer, TX_APP_MEM_POOL_SIZE) != TX_SUCCESS) {
        Error_Handler();
    }

    /*
     * Create other application threads, 3 times in this case
     *
     * Stack and thread handles are allocated with bytepool,
     * but cleanup process must be implemented to ensure no memory leakage
     */
    for (size_t i = 0; i < 3; ++i) {
        /**
         * Creation sequence
         *
         * - Allocate memory for thread handle
         * - Allocate memory for thread stack
         * - Call thread create function to register thread to the system
         */
        if (tx_byte_allocate(&tx_app_byte_pool, (void **)&t_handle, sizeof(*t_handle), TX_NO_WAIT) == TX_SUCCESS
                && tx_byte_allocate(&tx_app_byte_pool, &stack_mem, 512, TX_NO_WAIT) == TX_SUCCESS
                && tx_thread_create(t_handle, "app_thread", app_thread_func, i + 1,
                                    stack_mem, 512, 10, 0, TX_NO_TIME_SLICE, TX_AUTO_START) == TX_SUCCESS) {
            /* All good... */
        } else {
            /* Work on proper cleanup instead */
        }
    }
}

/**
 * \brief           Application thread function entry point
 *
 * It is used for several instances of the same thread
 * \param[in]       param: User parameter, index of thread in this particular case
 */
static void
app_thread_func(ULONG param) {
    debug_printf("[T%d] Thread enter\r\n", (int)param);
    tx_thread_sleep(1000 * param);
    debug_printf("[T%d] Thread after sleep\r\n", (int)param);

    /* Some threads will terminate, some will end by exiting the function call */
    if (param == 1) {
        /* Terminate yourself */
        debug_printf("[T%d] Thread terminating\r\n", (int)param);
        tx_thread_terminate(tx_thread_identify());
    }
    debug_printf("[T%d] Exiting the thread...\r\n", (int)param);

    /* Here we are suppose to exit from thread */
}
```

Is there:
- A way to set flag in the thread handle about dynamic allocation?
- A way to get a global system callback when we can really cleanup memory?

goldscott, 2022-01-10:
Hi @MaJerle - thanks for bringing up some very good issues. 

Your concept seems correct to me, but I didn't actually run it. 

Answers to your questions:

Yes, you should call tx_thread_delete to delete the thread. 

There is nothing in ThreadX that indicates whether the thread and its stack were dynamically allocated. You'll need to add this. 

In the default thread struct, there is no member/flag indicating dynamic allocation. You can use TX_THREAD_USER_EXTENSION in tx_port.h to add this yourself: https://github.com/azure-rtos/threadx/blob/b216ceb25e2b6ae1ea83c1ae91d114a6fc2fc4ec/ports/cortex_m4/gnu/inc/tx_port.h#L251

Likewise, you can use the TX_THREAD_DELETE_PORT_COMPLETION macro to do the memory cleanup. This may be more efficient than an idle thread that does the cleanup. ThreadX was designed to not have an idle thread consuming CPU & resources. Perhaps we should add actual user extensions to functions like delete, terminate, etc. I'll bring this up with the ThreadX team.

MaJerle, 2022-01-11:
For the moment it is done like this

```c
/* TX user extension */
#define TX_THREAD_USER_EXTENSION            UINT tx_thread_user_is_handle_alloc_dynamic; \
                                            UINT tx_thread_user_is_stack_alloc_dynamic;
```

Thread creation
```c
    /* Start app thread... */
    if (tx_byte_allocate(&tx_app_byte_pool, (void **)&t_handle, sizeof(*t_handle), TX_NO_WAIT) == TX_SUCCESS
            && tx_byte_allocate(&tx_app_byte_pool, &stack_mem, 4096, TX_NO_WAIT) == TX_SUCCESS
            && tx_thread_create(t_handle, "app_thread", app_thread_func, 0,
                                stack_mem, 4096, 10, 0, TX_NO_TIME_SLICE, TX_DONT_START) == TX_SUCCESS) {
        /* All good... */

        /* Set user fields since this is dynamic allocation */
        t_handle->tx_thread_user_is_handle_alloc_dynamic = 1;
        t_handle->tx_thread_user_is_stack_alloc_dynamic = 1;

        /* Start thread */
        tx_thread_resume(t_handle);
    }
```

Thread Idle module
```c
#include "tx_idle_thread.h"
#include "tx_api.h"
#include "tx_thread.h"
#include "mcu.h"

/* Idle thread uses pre-defined handle structure and statically defined stack memory */
DTCM_RAM static TX_THREAD idle_thread_handle;
DTCM_RAM static uint8_t idle_thread_stack[512];
static void idle_thread_func(ULONG param);

/**
 * \brief           Idle thread function entry
 * \param[in]       param: User argument, don't care
 */
static void
idle_thread_func(ULONG param) {
    TX_INTERRUPT_SAVE_AREA
    TX_THREAD* t;

    while (1) {
        /* Process interrupts and check if anything to do */
        TX_DISABLE

        /*
         * Check a linked list of all created threads in the system.
         * Process threads in terminated or completed state.
         *
         * These threads use dynamic memory allocation,
         * hence it is now good time to clean them up
         */
beg:
        t = _tx_thread_created_ptr;
        while (t != NULL) {
            if (t->tx_thread_id != TX_THREAD_ID) {
                break;
            }
            if (t->tx_thread_state == TX_TERMINATED
                    || t->tx_thread_state == TX_COMPLETED) {
                TX_RESTORE

                /* Do we need a call to delete? */
                tx_thread_delete(t);

                /* Cleanup resources */
                /* How should we determine if allocation was dynamic or static? */
                if (t->tx_thread_user_is_stack_alloc_dynamic) {
                    tx_byte_release(t->tx_thread_stack_start);
                }
                if (t->tx_thread_user_is_handle_alloc_dynamic) {
                    tx_byte_release(t);
                }

                TX_DISABLE

                /* Start from beginning since linked list changed now after delete operation */
                goto beg;
            }
            /* Find next thread */
            if (t->tx_thread_created_next != t
                    && t->tx_thread_created_next != _tx_thread_created_ptr) {
                t = t->tx_thread_created_next;
            } else {
                t = NULL;
            }
        }
        TX_RESTORE

        /* Yield thread */
        tx_thread_relinquish();
        tx_thread_sleep(1);
    }
}

/**
 * \brief           Start idle thread
 * \return          `1` if started, `0` otherwise
 */
uint8_t
tx_idle_thread_start(void) {
    /*
     * Create idle thread to monitor all other threads, having lowest priority
     *
     * Idle thread has lowest priority and static handle/stack objects
     */
    return tx_thread_create(&idle_thread_handle, "idle_thread", idle_thread_func, 0,
                     idle_thread_stack, sizeof(idle_thread_stack), TX_MAX_PRIORITIES - 1, 0, TX_NO_TIME_SLICE, TX_AUTO_START) == TX_SUCCESS;
    
}
```

I am not sure it is the right way of doing it.

goldscott, 2022-01-12:
Hi @MaJerle - your logic looks correct to me.

MaJerle, 2022-01-13:
Any chance you add this to ThreadX module as conditional compilation feature?

closed 2022-01-13 by MaJerle