eclipse-threadx/threadx #156: Does threadx suppport Cortex-A55?

chojpsh1, 2022-01-18:
Hello, I'm studying threadx nowadays.
I saw threadx supports the Cortex-Ax board. So, I wonder if this means that it also supports A55 board. If threadx supports, then how can I use it? Should I define some configurations for A55?

Thank you!

goldscott, 2022-01-19:
You can use the Cortex-A5x or Cortex-A35 ports.

chojpsh1, 2022-01-19:
So to sum up, I understand that threadx supports Cortex-A55 with using Cortex-A35 ports. right?
I would appreciate it if you could let me know there is anything I misunderstand.

goldscott, 2022-01-19:
Correct!

chojpsh1, 2022-01-19:
Okay, thank you for answer! :)

chojpsh1, 2022-01-19:
I have one more question. I know Cortex-A35 is ARMv8.1 and A55 is ARMv8.2, the extension version is different. So actually, I don't understand exactly how can threadx support A55 with A35 ports.
how threadx support A55 through A35 with different extension version?

goldscott, 2022-01-19:
Take the A35 port and build it for the A55. It will work. 
In a few more months, we will have official releases for more of the A processors.

chojpsh1, 2022-01-20:
Okay, I'll try it. Thank you!

chojpsh1, 2022-01-25:
May I know when more of the A processors release? I'd like to know the exact threadX's plan for A processors.

goldscott, 2022-01-26:
At the end of this week or next week we will be putting up a bunch more Cortex-A ports, including A55.

chojpsh1, 2022-01-26:
That's good news for me! :) Thank you

chojpsh1, 2022-01-28:
If A55 is supported, do I buy a license for A55 to use it? If so, I would like to ask you estimate the price of the license.

goldscott, 2022-01-31:
@chojpsh1 Cortex-A55 ports are now available: https://github.com/azure-rtos/threadx/tree/master/ports/cortex_a55

For licensing questions, please contact us here: https://azure-rtos.ms-iot-contact.com/ or here: https://azure-rtos.ms-iot-contact.com/sales/

closed 2022-01-31 by goldscott