eclipse-threadx/threadx #158: Cortex-M33 FPU register not saved/restored with GCC

Ralf264, 2022-01-21:
The code available in the Cortex-M33 files _tx_initialize_low_level.S_ and _tx_thread_schedule.S_ depend on macro ___ARM_PCS_VFP_ which is not available in gcc-10.3-2021.10. But gcc defines __ARM_FP that may be sufficient.

goldscott, 2022-01-21:
Thank you for bringing this to our attention. We will fix this in the next release.

Ralf264, 2022-01-21:
Forgot to mention that this probably also applies to the Cortex-M3, M4 and M7 ports. Since it's related, is the stack setup ok with FPU enabled?

goldscott, 2022-01-21:
@Ralf264 I will make the change to the M3/4/7 too. When did gnu make this change?

Ralf264, 2022-01-22:
@goldscott I have just learned that the presence of ___ARM_PCS_VFP_ depends on the _-mfloat-abi_ option:

- _-mfloat-abi=hard_ defines ___ARM_PCS_VFP_ and ___ARM_FP_
- _-mfloat-abi=softfp_ defines ___ARM_FP_ only
- _-mfloat-abi=soft_ both undefined

I've used softfp (for whatever reason) and the compiler was using the FPU registers in a thread as temporary storage.

The CMSIS header files use ___ARM_FP_ only...

goldscott, 2022-01-24:
Ah, thank you! I have changed our Cortex-M code to use __ARM_FP. It will be in the next release in about a week.

goldscott, 2022-01-31:
@Ralf264 this has been fixed in 6.1.10, which is now available. Thanks!

closed 2022-01-31 by goldscott