eclipse-threadx/threadx #160: Oversized shift in _tx_thread_smp_next_priority_find

uricohen1112, 2022-02-07:
When compiling with TX_MAX_PRIORITIES=32 for ARMv7-A processors, and there exists a process with priority 31, the function _tx_thread_smp_next_priority_find() gets called with priority=32, which then computes:
`local_priority_map = local_priority_map >> priority`
Since local_priority_map has type ULONG, it consists of 32 bits, therefore the shift is undefined.

Under `TX_MAX_PRIORITTIES > 32` ifdef there is another version of this function which validates `priority < TX_MAX_PRIORITIES` and otherwise returns TX_MAX_PRIORITIES, so maybe a similar check can be added to the other variant of this function.

goldscott, 2022-02-09:
Thank you for finding this. We are working on it and will have a fix in the next ThreadX release.

uricohen1112, 2022-02-13:
Thanks a lot!
I forgot to mention that this was found using UBsan (-fsanitize=undefined in gcc), so running any unit-tests with this tool might expose more such misuses, if there are any.

goldscott, 2022-04-26:
This was fixed in 6.1.11: https://github.com/azure-rtos/threadx/blob/master/common_smp/src/tx_thread_system_suspend.c

closed 2022-04-26 by goldscott