eclipse-threadx/threadx #161: SP select for armv8a profile when stack building

xjbwd, 2022-02-17:
For armv8a profile ports, when building stack in tx_thread_stack_build.S, the SPSR initial value is 0xC (program runs in highest privilege and EL3 defined). In this case, the program will use SP_EL0 as its SP when return from exception, but the SP_EL3 is expected.

SPSR description from armv8a arch spec:
> M[3:0], bits [3:0]
> AArch64 Exception level and selected Stack Pointer.
> 0b0000 EL0t.
> 0b0100 EL1t.
> 0b0101 EL1h.
> 0b1000 EL2t.
> 0b1001 EL2h.
> 0b1100 EL3t.
> 0b1101 EL3h.
> 
> The selected stack pointer can be indicated by a suffix to the Exception level:
> t Indicates use of the SP_EL0 stack pointer.
> h Indicates use of the SP_ELx stack pointer.

If the EL3 used, 0xD is the correct initial value for SPSR?
Is my understanding right?

Thanks.

goldscott, 2022-02-17:
I think you are correct. Please take a look at this port (Cortex-A5x) and you'll see that LSB is set: https://github.com/azure-rtos/threadx/blob/master/ports/cortex_a5x/ac6/src/tx_thread_stack_build.S
We will have to fix that for the others.

xjbwd, 2022-02-18:
Get it, thanks.

Another question, what platform are you guys using for smp of v8a ？

I am testing dual a55 core thread scheduling on a HW emulator, Each thread is simple just sleep for a while and wake up. 
The thread scheduling is abnormal some times later, _tx_thread_preempt_disable flag is always 1 and the tx_thread_sleep return error. Another core can not get the lock.

We are following this issue further, is there any suggestion ?

goldscott, 2022-02-18:
I believe we have done most of the development and testing in the ARM FVP simulator as well as 4-core Cortex-A53 hardware. 

What emulator are you using? We would like to try to reproduce what you are observing.

xjbwd, 2022-02-21:
If TX_NOT_INTERRUPTABLE defined, the issue seems to be solved.

goldscott, 2022-04-15:
@xjbwd is there a reason you are running in EL3 and not EL1? I believe EL1 is the recommendation for OSes (secure monitors should run in EL3).

xjbwd, 2022-04-22:
I'll give a try if there are resources

goldscott, 2022-05-09:
Hi @xjbwd - some updates for you: our Cortex-A5x port was written first and set up to use EL3. All of the other ports (A53, A55, A57, all the A3x, A6x, A7x, etc) default to EL1. ARM (and Microsoft) strongly suggest you run in EL1. You may need to modify your startup code to set this up properly. The example we provide for the A55 does this: https://github.com/azure-rtos/threadx/tree/master/ports_smp/cortex_a55_smp/gnu/example_build
If you are still having problems, let us know, otherwise I will close this ticket soon.

xjbwd, 2022-06-24:
Get it, thanks.

closed 2022-06-24 by xjbwd