eclipse-threadx/threadx #163: tx_user.h tweaks in Cortex-M GNU assembly files

udoe, 2022-02-28:
Hi,

I'm referring to the assembly files in ports/cortex_m7/gnu/src (The issue should be present in other cortex_m/gnu ports as well but I haven't checked this).

Shouldn't all those .S files #include "tx_user.h" at the top so that tweaks such as TX_ENABLE_WFI or TX_EXECUTION_PROFILE_ENABLE made therein have an effect on the assembly code? Without that include statement I need to duplicate such defines in the makefile which is error-prone.

Thanks
Udo

goldscott, 2022-02-28:
We have been discussing this internally and hope to make some improvements soon.

TiejunMS, 2023-03-14:
Included tx_user.h in GNU port 
https://github.com/azure-rtos/threadx/commit/2aa19f3de0b6cf09bb9dca78c8a53fe337b8f0f7#diff-9c9ffe27e291d6c526fad16b85f2d486b27cdde739065e247f5d3f2198247961

closed 2023-03-14 by TiejunMS