eclipse-threadx/threadx #164: TX_EXECUTION_TIME_SOURCE definition lacks volatile qualifier

udoe, 2022-02-28:
Hi,

The TX_EXECUTION_TIME_SOURCE example for Cortex-M in tx_execution_profile.h
`#define TX_EXECUTION_TIME_SOURCE         (EXECUTION_TIME_SOURCE_TYPE) *((ULONG *) 0xE0001004)
`
should read
`#define TX_EXECUTION_TIME_SOURCE         (EXECUTION_TIME_SOURCE_TYPE) *((volatile ULONG *) 0xE0001004)
`

To prevent the compiler from eliminating, duplicating, or moving instructions that affect registers, any memory-mapped register address should be marked volatile. See also Arm's DWT_Type etc. definitions in core_cm7.h and friends.

Udo

goldscott, 2022-02-28:
Hi Udo, we will add the 'volatile' keyword; this will be fixed in the next release.

goldscott, 2022-04-26:
This has been fixed in 6.1.11: https://github.com/azure-rtos/threadx/blob/master/utility/execution_profile_kit/tx_execution_profile.h

closed 2022-04-26 by goldscott