eclipse-threadx/threadx #165: TX_EXECUTION_MAX_TIME_SOURCE wrap-around handling in tx_execution_profile.c

udoe, 2022-02-28:
Hello,

Affected functions:
_tx_execution_thread_enter
_tx_execution_thread_exit
_tx_execution_isr_enter
_tx_execution_isr_exit

In the wrap around case, the ticks elapsed are calculated like this:
` delta_time =  (EXECUTION_TIME) (current_time + (((EXECUTION_TIME_SOURCE_TYPE) TX_EXECUTION_MAX_TIME_SOURCE) - last_start_time));
`

This should be:
`delta_time =  (EXECUTION_TIME) (current_time + ((((EXECUTION_TIME_SOURCE_TYPE) TX_EXECUTION_MAX_TIME_SOURCE) + 1) - last_start_time));
`

Otherwise there is an error of one tick.

Consider the corner case where last_start_time=0xFFFFFFFF and current_time=0x00000000. This leads to delta_time=0 which is not correct. The fixed line above produces delta_time=1.

Udo

goldscott, 2022-02-28:
Hello Udo,

Thank you for bringng this up. We will fix it in the next release.

goldscott, 2022-04-26:
This has been fixed in 6.1.11: https://github.com/azure-rtos/threadx/blob/master/utility/execution_profile_kit/tx_execution_profile.c

closed 2022-04-26 by goldscott