eclipse-threadx/threadx #172: low power on ARM: LR should be saved _tx_thread_schedule

foldl, 2022-03-10:
Task this as an example, after `BL tx_low_power_exit`, `LR` is overwritten. Later, when a new task is ready to run, `BX lr` would not work.

https://github.com/azure-rtos/threadx/blob/master/ports/cortex_m0/keil/src/tx_thread_schedule.s#L264

```asm
    IF :DEF:TX_LOW_POWER
    PUSH    {r0-r3}
    BL      tx_low_power_exit                       ; Exit low power mode
    POP     {r0-r3}
    ENDIF
```

goldscott, 2022-03-10:
Yep, you're right. We'll fix this in the next release.

goldscott, 2022-03-10:
It doesn't matter what LR is here. When a thread is scheduled, it's LR will be restored.

foldl, 2022-03-11:
Yes, `LR` is restored.

I would like to provide more information on why I have created this issue.

In `TX_LOW_POWER_USER_ENTER`, MCU is powered off while RAM is retention-ed. After `tx_low_power_next_expiration`,
MCU is powered on again, I try to resume the execution:

```c
void low_power_user_enter()
{
    power off;
restore_here:  // after MCU powered on again (Thread mode), jump to restore_here
    restore context;
}
```

This works for RTOSes that have a dedicated idle task. While in ThreadX, `low_power_user_enter` is called in PendSV, it needs
extra tinkering to "resume" the PendSV ISR.

foldl, 2022-03-11:
Create an idle task and my solution works. Suggest adding documentation on this, warnings and tips.

foldl, 2022-03-16:
To resume/return to the PendSV ISR, I think we can:

1. After MCU powered on again from low power mode, set up a flag and trigger PendSV;
2. In PendSV, when the flag is set, jump to `restore_here`.

This solution is more complex than using an idle task, so I gave it up and created a dedicated idle task to do the work, :)

foldl, 2022-03-17:
Here is what I am doing before creating a dedicated idle task (ARM Cortex-M is assumed):

```c
reset_handler()
{
    if (wakeup from low power) jump to restore_here;
    ...
}

void low_power_user_enter()
{
    power off;
restore_here:  // after MCU powered on again (Thread mode), jump to restore_here
    restore context;
}
```

The concept here is to let caller of `low_power_user_enter` not feel anything about the low power procedure. 

The above low power procedure has been ported to quite some RTOSes. When porting to ThreadX, I gradually found that since `low_power_user_enter` is called in PendSV, it won't work.

goldscott, 2022-03-17:
Ah, this is where my confusion is coming from. Why is your wake-up from low power happening in the reset_handler and not resuming execution in the low_power_user_enter function? The WFI/WFE instructions should resume execution at the next instruction (or in an ISR), not in the reset_handler...

foldl, 2022-03-18:
Because in my case, MCU is totally powered off in low power mode. A little hardware component will turn on MCU again later.

goldscott, 2022-07-14:
Hi @foldl - you do _not_ have to make the calls to ```tx_low_power_enter``` and ```tx_low_power_exit``` in the scheduler like we do for Cortex-M. You can call these functions elsewhere if it suits your application (for example, in an idle thread, or call ```tx_low_power_exit``` in your reset vector code when the processor wakes from sleep).

closed 2022-07-15 by goldscott