eclipse-threadx/threadx #173: the posix_arrange_msg function crashes.

boria8, 2022-03-10:
it crashes when more than 1 message of the same priority present in the messageQ.

solution:
To verify temp_q pointer is not null before Swap the messages:
 if(temp_q)
{
    /* Swap the messages.  */
    for ( msg = 0; msg < 4; msg++)
    {

        swap = *temp_q;
        *temp_q = *Qread;
        *Qread = swap;
        temp_q++;
        Qread++;
    }
}

goldscott, 2022-03-17:
Do you have a test case for this? I am putting two and three messages of the same priority in the queue and not seeing any crash.

goldscott, 2022-10-28:
@boria8 - as of release 6.2.0, this issue has been fixed: https://github.com/azure-rtos/threadx/blob/master/utility/rtos_compatibility_layers/posix/px_mq_arrange_msg.c

closed 2022-10-28 by goldscott