eclipse-threadx/threadx #178: Use of queues for single chars as individual messages

XaviGNRGY, 2022-03-16:
Hi, we are having a problem regarding to the size of queue's messages.

We are moving our project from freeRTOS to threadX. In freeRTOS we used to create queues dinamically, assigning the individual message size as one wanted to use (from 1 byte to whatever).

Now trying to do the same with ThreadX we found 2 problems. The first one is solved as is explained in the documentation, how to proceed when the mesage size exceeds 16 words length, passing pointers instead of the message struct itself... Fine! 👍 

Now, what happens when the message unit that you want to send to a queue is just a simple 8bits char ? we will be wasting 3/4 of the memory content. That means if I want to implement a queue to hold 1024 characters until they can be processed, we need to declare a queue of 4096 Bytes, resulting in 3072 bytes completely useless.

Is there any solution for this particlar case?

goldscott, 2022-03-17:
True, queuing single bytes is wasteful in this scheme. Are you able to wait until you have 4 bytes to queue and pack them into a single message?

XaviGNRGY, 2022-03-18:
Hi Scott,

Indeed we have considered this option, but  the algorithm that we use requires processing every single character right after its reception.
Also, we would like to keep the project as closest as possible to the one that we have developed with freeRTOS, just in case we needed to move backwards and retake it.

TiejunMS, 2023-03-14:
Another solution is to use semaphore and ring buffer for single byte buffer. Closing.

closed 2023-03-14 by TiejunMS