eclipse-threadx/threadx #180: Could I know if there is any settings that I missed for executing sample threadx?

chojpsh1, 2022-03-30:
Hello, I have a question about executing sample threadx on Arm Development Studio IDE with FVP_Cortex_A55.

I opened the example_build project which is located at ports_smp/cortex_a55_smp/gnu/ and built the tx and sample_threadx project. So, I saw the sample_threadx.axf file was saved in Debug directory.
I created the debug model connection which's target is FVP_Cortex_A55x1 and connected to sample_threadx.axf but, any threads didn't execute and loop in _tx_initialize_kernel_enter. 

Is there a process or settings that I missed in executing the thread?

carll00226, 2022-03-30:
your target is cortex_a55*1, it's single core, not smp, you should port form ports/, not ports_smp/.

chojpsh1, 2022-03-30:
Thank you for reply @carll00226 :)
I modify the target FVP_cortex_a55*4 and check the TX_THREAD_SMP_MAX_CORES is 4.
But, threads still don't execute (System Timer Thread's state : SUSPENDED and thread 0~3 state : READY) and loop in _tx_initialize_kerenl_enter.

carll00226, 2022-03-31:
I think it is a coherence issue，core0 is waiting for other cores to be ready. you can check your setting about cache, mmu, scu...which do the coherence  job.

chojpsh1, 2022-04-01:
It's resolved by defining TX_ARMV8_2

closed 2022-04-01 by chojpsh1