eclipse-threadx/threadx #183: Green Hills Compiler support for ARM Cortex M33 missing but present in M4

ascheurer, 2022-04-08:
ports/cortex_m4 supports the Green Hills compiler (__ghs__) however the cortex_m33 supports ac3, gnu and iar only. Is there a plan in place to support Green Hills with the m33? Currently using the cortex_m4 but planning a transition to the m33 and would like to understand if the m33 with the ghs compiler is going to supported. 

Often there is a transition from the M4 to the M33 because of trustzone supported in the M33 but not in the M4 - memory protection is supported in the M4 but not the trust zone. 
Is using Green Hills compiler, probe going to be supported in the M33?

ascheurer, 2022-04-08:
Currently using gnu and keil compiler - keil supports the newer arm development studio (gnu flag compatible) but not sure of the rest. It is a major upgrade from the Keil v5 MDK which uses armcc instead of armclang.

goldscott, 2022-04-08:
Hi @ascheurer - can you please contact azure-rtos-support@microsoft.com in regard to the GHS compiler. We have not been regularly updating the GHS ports.

I was under the impression the armcc compiler (version 5) did not support the trustzone architectures. Our [ARM port for the M33](https://github.com/azure-rtos/threadx/tree/master/ports/cortex_m33/ac6) uses ARM Compiler 6. This compiler can be used with Keil IDE (in fact, the sample project we ship uses the Keil IDE). The ARM Compiler v6 now comes with Keil IDE, see link for more: https://www2.keil.com/mdk5.

goldscott, 2022-04-25:
Closing for now. Feel free to email us for more info.

closed 2022-04-25 by goldscott