eclipse-threadx/threadx #184: Floating point build for threadx

NaikPrasannaSe, 2022-04-19:
Hi,

I am evaluating Threadx rtos on LPC54608J512BD208 microcontroller (Cortex M4).
I am using MCUXpresso IDE v10.5 of NXP.
I have built the Threadx library from the batch file given in example folder of gnu/cortex M4.
During porting, I am facing issues related  to floating point registers.
The following error is displayed by the compiler when i link the Threadx library:
**error: lpcxpresso54608_gpio_led_output2.axf uses VFP register arguments, C:\Users\20071270\Desktop\Threadx RTOS Workspace\LPC546 Workspace\lpcxpresso54608_gpio_led_output2\azure-rtos\binary\threadx\cortex_m4\mcux\libthreadx_sp.a(tx_block_allocate.o) does not**

On researching about this error, i found that the Threadx library generated by build file doesn't floating point engine and hence the compiler gives error since the application code uses floating point engine.
I couldn't find much data about threadx porting for cortex M4 online, hence raising the query here.

Thanks and best regards,
Prasanna

andrejm-rtos, 2022-04-19:
Hi Prasanna,

The error message is not complete in your post. I cannot tell for sure what the problem is. Assuming it is about floating-point support, like you say, ThreadX for Cortex M4 supports floating point without any additional changes. 

The ThreadX source code assumes the symbol __ARM_FP is defined by the compiler and assembler, that it is the case by default, but you should check your environment. 

There is a bit more information on the readme, section 7 here:
https://github.com/azure-rtos/threadx/blob/master/ports/cortex_m4/gnu/readme_threadx.txt

It is not clear from your original post if you put together by yourself an example or you used one of the examples provided by NXP. For evaluating ThreadX with an NXP LPC chip using MCUXpresso, I would recommend using the examples provided by NXP.

goldscott, 2022-04-26:
@NaikPrasannaSe were you able to rebuild with VFP support enabled? Is this still an issue?

NaikPrasannaSe, 2022-04-27:
Hi Scott,

I am still in communication with NXP team. We haven't found the solution yet.

Thanks,
Prasanna

matiss, 2022-05-05:
Hi Prasanna,

I had the same error as you did. The documentation on VFP/FPU is a bit poor.
What I did is I modified the `cortex_m4.cmake` file to add correct compiler flags:
```set(VFP_FLAGS "-mfpu=fpv4-sp-d16 -mfloat-abi=hard")```

The VFP_FLAGS are blank by default. Make sure to use correct VFP flags for your MCU.

Then I re-generated cmake files and added `#define TX_ENABLE_FPU_SUPPORT` define to the `tx_user.h` file.

goldscott, 2022-05-05:
@matiss provided a great explanation. I will close this ticket now. Feel free to reopen if you have more questions.

NaikPrasannaSe, 2022-05-12:
Hi Scott/Matiss,

I am facing a problem while building threadx library through cmake:

1) I have used cmake (gui), Ninja and GCC Compilers for ARM-NONE-EABI downloaded from the links given in the main page of threadx github.
2) I have added Ninja to the environment path
3) I have done following settings in cmake (gui)
        a) Where is the source code: C:/Users/Lab/Downloads/threadx-master (1)/threadx-master/ports/cortex_m4/gnu
         b) Where to build the binaries: C:/Users/Lab/Downloads/New folder
         c) Configure: 
                            1.  Specify generator for this project: Ninja
                            2.   Specify toll chain for cross compilation: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gcc.exe
           
4)  I am getting following error in cmake:
 
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gcc.exe:1:
  Parse error.  Expected a command name, got unquoted argument with text
  "MZ�".
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt

CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.23)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring incomplete, errors occurred!

Please help in resolving this problem as i couldn't find any solution on the internet.

Thanks,
Prasanna

matiss, 2022-05-12:
I would suggest to use command line cmake commands as per documentation and not the GUI. Also you would be better off to ask this kind of questions in the official [ThreadX Q&A](https://docs.microsoft.com/en-us/answers/topics/azure-rtos.html).

goldscott, 2022-05-12:
@NaikPrasannaSe you can ask these questions here instead of Q&A. I am not very knowledgeable about cmake, so I will get somebody who knows more to help answer. Thanks!

goldscott, 2022-07-15:
I am closing this issue. If there are further questions, please re-open.

closed 2022-07-15 by goldscott