eclipse-threadx/threadx #186: Incorrect preprocessor macro in tx_thread_smp_protect.S

vks3, 2022-04-20:
In tx_thread_smp_protect.S, for the a7-smp port (probably in others too), the following lines have an incorrect preprocessor leading to some critical code being excluded and leading to deadlocks and random crashes.

Line99: 
```
#ifdef TX_ENABLE_FIQ_SUPPORT
    CPSID   if                                  @ Disable IRQ and FIQ interrupts
    _**ELSE**_
    CPSID   i                                   @ Disable IRQ interrupts
#endif
```
Instead of #else, ELSE is used which is ignored by the compiler if TX_ENABLE_FIQ_SUPPORT is not set.

The same issue is repeated at  Line 250 onwards.

This at least solved my present issue of CPU getting deadlocked and no threads being scheduled (issue #179).

goldscott, 2022-04-20:
Hi @vks3 - thank you for finding this! We have ThreadX version 6.1.11 coming out next week and can't make any more changes to it right now, so this will be fixed in 6.1.12.

goldscott, 2022-04-20:
Update: it looks like this is only in the Cortex-A7 SMP file that you mention. I did not find it anywhere else. I'll keep this ticket open and close it once version 6.1.12 has been released.

goldscott, 2022-07-28:
This was fixed in 6.1.12 release.

https://github.com/azure-rtos/threadx/blob/8c3c08f10826c4d5332a6d8e2fdf014b793a456a/ports_smp/cortex_a7_smp/gnu/src/tx_thread_smp_protect.S#L103

https://github.com/azure-rtos/threadx/blob/8c3c08f10826c4d5332a6d8e2fdf014b793a456a/ports_smp/cortex_a7_smp/gnu/src/tx_thread_smp_protect.S#L254

closed 2022-07-28 by goldscott