eclipse-threadx/threadx #189: tx_time_get() always return 0

chojpsh1, 2022-04-29:
Hi, I run the threadx on FVP cortex A55 8 cores and I want to measure the running time with 'tx_time_get' function.
So, I called the 'tx_time_get' at entry of my function and the end of my function but, it always return 0.
I would like to know if there is any necessary work (ex. setting configuration or call some function before tx_time_get ...) for using tx_time_get().

Thank you!

goldscott, 2022-04-29:
It sounds like you don't have a hardware timer set up to generate periodic interrupts for ThreadX. Do you have a timer set up?

chojpsh1, 2022-05-02:
I'm sorry but I don't understand exactly about hardware timer. Does it mean external timer?
For the time beginning, we measured the time by reading cntvct_el0 of timer register instead of using tx_time_get().

goldscott, 2022-05-03:
You need a hardware timer, it doesn't matter if it is internal to the chip (like the SysTick timer for Cortex-M, or an external timer generating a periodic interrupt). You need to call tx_timer_interrupt in the timer ISR. More info here: https://docs.microsoft.com/en-us/azure/rtos/threadx/chapter2#target-considerations
It sounds like you are using something like AArch64. Check out our example here: https://github.com/azure-rtos/threadx/blob/e11c166a4045d80ea74fe917e5608474754f5067/ports/cortex_a53/gnu/example_build/sample_threadx/timer_interrupts.c#L90

chojpsh1, 2022-05-09:
Thank you @goldscott ! I did it using the example as your comment.
In addition, I would like to know if there is a way to measure memory increment for thread. (I want to know how much size of memory increased if 4KB-thread created.) 
I know the stack size is 4096 bytes(in 4KB-thread), but I wonder is there any additional structure usage's size for thread.

goldscott, 2022-05-09:
I'm not sure I understand your question. The size of a thread is `size_of(TX_THREAD)`. The stack is separate from the thread struct. If you need more information, please create a separate ticket.

closed 2022-05-09 by goldscott