eclipse-threadx/threadx #191: Reads random memory if object_name is NULL

fschwa, 2022-05-16:
https://github.com/azure-rtos/threadx/blob/e11c166a4045d80ea74fe917e5608474754f5067/common/src/tx_trace_object_register.c#L219

If you create an object (e.g. with tx_mutex_create) with out giving a name (NULL), the `for` loop referenced above reads random memory, which succeeds sometimes and does weird stuff otherwise. Should it maybe be mentioned in the API docs that NULL is not a good idea or maybe even caught within the function?

goldscott, 2022-05-17:
Good find! We will fix this in our next release (6.1.12, coming out in a couple months). I'll keep this ticket open until the release has been made.

goldscott, 2022-07-28:
This has been fixed in the 6.1.12 release
https://github.com/azure-rtos/threadx/commit/8c3c08f10826c4d5332a6d8e2fdf014b793a456a#diff-574e7d43192b206e6ab63351b0f05a0105174151458cb509f4f6b5eeccff137b

closed 2022-07-28 by goldscott