eclipse-threadx/threadx #192: Why return TX_NO_INSTANCE when semaphore timeout?

beny-p, 2022-05-23:
Why return TX_NO_INSTANCE when semaphore timeout?
There are 3 return from tx_semaphore_get:
1, return for preempt_disable is set;
2, return for wait time out;
3, return for not wait;
However, they return the same value TX_NO_INSTANCE.
We can not do distinguish one reason from another.
Is there some using advice? or would you using different return value, such as SUCCESS, or WAIT_ABORTED?

goldscott, 2022-05-23:
https://docs.microsoft.com/en-us/azure/rtos/threadx/chapter4#tx_semaphore_get

TX_NO_INSTANCE means "Service was unable to retrieve an instance of the counting semaphore (semaphore count is zero within the specified time to wait)."

There is no reason to specify a different return status for a wait timeout of 0 (i.e. not waiting) versus waiting some number of ticks: the result is still the same - the semaphore is not available.

beny-p, 2022-05-25:
Got it. Thanks.

closed 2022-05-25 by beny-p