eclipse-threadx/threadx #193: nanosleep  in posix layer may return error

Albert-Coder, 2022-06-07:
nanosleep in px_nanosleep.c  in posix layer , when req->tv_nsec = 0 but req->tv_sec != 0, it may return EINVAL;
I found the description 
"/* Check for valid inputs */
    /* The nanosecond value must be greater than zero or less than  1 000 million. */ "
this may be diffrent from posix standard IEEE Std 1003.1, 2004 Edition, I picked them at below

NAME
nanosleep - high resolution sleep (REALTIME)
SYNOPSIS
[[TMR](javascript:open_code('TMR'))] [Option Start] #include <[time.h](https://github.com/azure-rtos/threadx/basedefs/time.h.html)>

int nanosleep(const struct timespec *rqtp, struct timespec *rmtp); [Option End]

ERRORS
The nanosleep() function shall fail if:

[EINTR]
The nanosleep() function was interrupted by a signal.
[EINVAL]
**The rqtp argument specified a nanosecond value less than zero or greater than or equal to 1000 million.**

These two differnce is whether or not check (nanosecond value equal 0 ) and (nanosecond value equal  1000 million).

goldscott, 2022-08-24:
This has been fixed internally and will be available in our next release.

goldscott, 2022-10-28:
This has been fixed here: https://github.com/azure-rtos/threadx/blob/master/utility/rtos_compatibility_layers/posix/px_nanosleep.c

closed 2022-10-28 by goldscott