eclipse-threadx/threadx #194: Wrong SPSR value in tx_thread_stack_build.S for Cortex A53 ports.

rodfrazer, 2022-06-14:
I've been evaluating the Cortex A53 port for both the SMP and non-SMP ports and it appears to me that the SPSR value pushed into the stack frame does not set bit 0 which causes the stack pointer to select SP_EL0 instead of keeping the stack pointer from the respective execution level that the code is running in.  The rest of the code does not seem to expect this behavior.  I've noticed that in the cortex_a5x port this issue has already been resolved.  Can the cortex_a5x implementation be copied over to the Cortex A53 ports?

Here are the Cortex A53 code sections that I'm aware of that do it wrong:
[ports_smp/cortex_a53_smp/gnu/src/tx_thread_stack_build.S#L140](https://github.com/azure-rtos/threadx/blob/83b57acde9dddf1c9468f0f3f0bed7a764f73c85/ports_smp/cortex_a53_smp/gnu/src/tx_thread_stack_build.S#L140)
[ports/cortex_a53/gnu/src/tx_thread_stack_build.S#L140](https://github.com/azure-rtos/threadx/blob/83b57acde9dddf1c9468f0f3f0bed7a764f73c85/ports/cortex_a53/gnu/src/tx_thread_stack_build.S#L140)

And this is the cortex_a5x code section that does this correctly:
[ports/cortex_a5x/ac6/src/tx_thread_stack_build.S#L150](https://github.com/azure-rtos/threadx/blob/83b57acde9dddf1c9468f0f3f0bed7a764f73c85/ports/cortex_a5x/ac6/src/tx_thread_stack_build.S#L150)

rodfrazer, 2022-06-14:
I did see that other issue but I don't see the code from the cortex-a5x port being replicated into any of the other ports yet to address the problem. Is that still planned to happen in a future release?

goldscott, 2022-06-14:
I am not sure the a5x code should be propagated to the other ports. What EL are you operating in?

rodfrazer, 2022-06-14:
I happen to be running in EL2 mode, but it seems to me like this code breaks in EL1, EL2 or EL3 mode.  With the M[0] bit clear in the SPSR the ERET that executes to enter the thread context is going to implement the SP_EL0 stack pointer regardless of what context the scheduler may be operating in but I see no code in the ports that seems to contemplate that.  There doesn't appear to be any code configuring and maintaining SP_EL0 in the scheduler code, so my assumption is that the scheduler expects the stack pointer to remain on the SP_ELx stack pointer.

On my target the SP_EL0 stack pointer is pointing to 0x0 as I never expect it to be used, and accesses on my target to address 0 - 1M will cause an access fault.  So when I start the RTOS and the _tx_thread_schedule_loop executes the ERET to start my first thread, the first instruction at _tx_thread_shell_entry pushes to the stack and my target experiences the fault due to accessing location 0x0 from the SP_EL0 stack pointer.  When I change the code to configure an SPSR with the M[0] bit set, then I get normal expected operation as I start the kernel and the SP_ELx stack pointer is allowed to persist through the ERET instruction.

fdesbiens, 2026-09-08:
Hi @rodfrazer, apologies for the very long silence on this one, and thanks for the detailed analysis.

I have gone through the ARMv8-A ports on `dev` and I do not think this is a bug. There are two different, internally consistent designs in the tree, and the SPSR value is only meaningful in the context of the one the port uses.

The `cortex_a5x` port was written first and targets EL3. It never touches `SPSel`, so it stays on `SP_ELx` throughout and its `0x5`/`0x9`/`0xD` (`ELxh`) values are the correct ones for it.

All the later ARMv8-A ports — A34, A35, A53, A55, A57, the A6x and A7x families, plus the SMP and module variants — are built on the opposite convention: the kernel runs with `SPSel = 0` and uses `SP_EL0` for both thread stacks and the system stack. `SP_ELx` is only a transient exception-entry stack. You can see the contract in the supplied example code: `startup.S` sets the handler stack, then executes `MSR SPSel, #0` before setting the real stack, and `irqFirstLevelHandler` in `vectors.S` executes `MSR SPSel, 0` before branching to `_tx_thread_context_save`. Consistently with that, `_tx_thread_schedule` restores the thread stack with a plain `MOV sp, x4`, which writes `SP_EL0`, and the `ERET` with `M[0]` clear keeps the processor on `SP_EL0`.

So the `ELxt` value is correct for those ports, and propagating the a5x value into them without also reworking the startup code, the vector table and the scheduler would break them.

What you are hitting is that your environment runs with `SPSel = 1` and leaves `SP_EL0` pointing at 0, so the `ERET` into the first thread switches to an unmapped stack. The supported fix is to follow the example startup sequence: initialise `SP_EL0` to your stack and execute `MSR SPSel, #0` before entering ThreadX, and make sure your IRQ entry path switches to `SP_EL0` before calling `_tx_thread_context_save`.

I agree this is far too easy to trip over, and it has now come up twice (see also #161), so I am closing this issue as answered and filing #703 to get the `SPSel = 0` requirement documented properly for the ARMv8-A ports.

closed 2026-09-08 by fdesbiens