eclipse-threadx/threadx #195: 使用mutex跑死

chengshuihang, 2022-06-21:
您好，我在使用mutex的过程中，出现了跑死的问题。
下面是跑死时候的资源截图
[image]
下面截图是mutex资源的截图
[image]
下面是我现在使用mutex的代码
`/*
 * fx_stm32_sdmmc1_driver.c
 *
 *  Created on: Apr 6, 2022
 *      Author: chengshuihang
 */

/* Include necessary system files.  */

#include "tx_api.h"
#include "fx_api.h"
#include "fx_user.h"

//#define SD_DETECT_PIN         CardCD_Pin
//#define SD_DETECT_GPIO_PORT   CardCD_GPIO_Port

/* Define semaphore for media transfer.  */
static TX_SEMAPHORE semaphore_transfer;
static TX_MUTEX mutex_transfer;
static         	__ALIGNED(4) uint8_t rx_dma_buff[512];
static         	__ALIGNED(4) uint8_t tx_dma_buff[512];

VOID  _fx_stm32_sdmmc1_driver(FX_MEDIA *media_ptr);
static void wait_DMA_read_complete();
static void wait_DMA_write_complete();
static void wait_transfer_ok();

UINT  _fx_partition_offset_calculate(void  *partition_sector, UINT partition,
                                     ULONG *partition_start, ULONG *partition_size);

void BSP_SDMMC1_WriteCpltCallback(void)
{
    /* Put semaphore for transfer completion.  */
    tx_semaphore_put(&semaphore_transfer);
}

/**
  * @brief BSP Rx Transfer completed callbacks
  * @param None
  * @retval None
  */
void BSP_SDMMC1_ReadCpltCallback(void)
{
    /* Put semaphore for transfer completion.  */
    tx_semaphore_put(&semaphore_transfer);
}

int BSP_SDMMC1_Init(void)
{
	UINT status;

	status = tx_semaphore_create(&semaphore_transfer, "SDMMC transfer semaphore", 0);
	if(status != TX_SUCCESS){
		return (0 - (int)status);
    }

	status = tx_mutex_create(&mutex_transfer, "SDMMC transfer mutex", TX_NO_INHERIT);
    if(status != TX_SUCCESS){
    	return (0 - (int)status);
    }

	return 0;
}

[... 460 lines cut ...]
            /* Check status of SDMMC Write.  */
            if (SDMMC_Status == HAL_OK){
                wait_DMA_write_complete();
                /* Successful driver request.  */
                media_ptr->fx_media_driver_status = FX_SUCCESS;
            }
            else{
                /* Unsuccessful driver request.  */
                media_ptr->fx_media_driver_status = FX_IO_ERROR;
            }
            wait_transfer_unlock();
            break ;
        }

        default:{
            /* Invalid driver request.  */
            media_ptr->fx_media_driver_status = FX_IO_ERROR;
            break;
        }
    }
}
`
在这里添加mutex的原因，是因为emmc按照大小被分成了几个FX_MEDIA的实例，在不同线程中使用，为了保证数据正确，所以添加了mutex。我mutex操作有问题吗？
下面是media_ptr -> fx_media_driver_request的值
[image]

TiejunMS, 2022-06-22:
@chengshuihang , thanks for reporting issue. It looks like the issue is related to FileX. Could you move the issue to FileX repo?

chengshuihang, 2022-06-22:
我尝试了一下，将互斥锁修改为二值信号量，它将不会跑死了，目前运行的很正常，系统稳定性我还在测试。这种情况issue还需要迁移吗？

TiejunMS, 2022-06-22:
Glad to know you have solved the issue. In that case, I will just close it and there is no need to migrate to FileX issue. If you have more findings, feel free to open a new issue in FileX repository.

closed 2022-06-22 by TiejunMS

chengshuihang, 2022-06-22:
好的