eclipse-threadx/threadx #199: How to use threadx on the basis of freertos

xiaguangbo, 2022-07-10:
Like threadx provides freertos api

goldscott, 2022-07-10:
We have a FreeRTOS conversion layer here: https://github.com/azure-rtos/threadx/tree/master/utility/rtos_compatibility_layers/FreeRTOS

xiaguangbo, 2022-07-11:
@goldscott I wasn't accurate enough. It should be like linux port or win port, provide a freertos port

goldscott, 2022-07-11:
I don't understand what you are asking for. ThreadX is a different operating system than FreeRTOS. In my previous comment, I linked to a conversion layer that allows you to use a FreeRTOS application with ThreadX as the actual RTOS.

xiaguangbo, 2022-07-12:
@goldscott I want to use threadx on esp32s2. esp32s2 is developed using esp-idf, and some functions are bound to freertos provided by esp-idf, I want to use threadx on top of this.
Although ports/xtensa is provided, this is not the best solution for esp32s2.
What I expect like is ports/linux or ports/win32 to simulate the basic functions of threadx with the system's api. In this way, the original function can not be affected to the greatest extent, and threadx can also be used.

goldscott, 2022-07-12:
The only ThreadX port we have for esp32 is the xtensa port, so you'd have to use that. 

I'm still not sure what you are trying to do. You are trying to simulate ThreadX with FreeRTOS? Or a conversion layer that converts ThreadX to underlying FreeRTOS?

xiaguangbo, 2022-07-12:
@goldscott 
Add a threadx/ports/freertos
If the role of threadx/ports/linux is to make threadx work on linux, then threadx/ports/freertos is to make threadx work on freertos
[image]

goldscott, 2022-07-12:
ThreadX on Linux is a special port compared to the rest of the ports. ThreadX actually uses posix pthreads for each ThreadX thread, and the ThreadX scheduler runs in a high priority pthread. This scheme will not work on FreeRTOS.

closed 2022-07-12 by goldscott