eclipse-threadx/threadx #200: Cortex-M7 BASEPRI does not work properly

MaJerle, 2022-07-10:
I have noticed, when:

- `TX_PORT_USE_BASEPRI` compilation flag is enabled
- `TX_PORT_BASEPRI` set to `5` and all interrupts that use ThreadX API have the same (or higher number, lower irq) pre-emption IRQ

USBX device CDC mode stops working. Port for STM32 never gets properly transfer interrupt.
Disabling `TX_PORT_USE_BASEPRI` leads to proper operation.

I'm still looking for an issue conclusion, but so far I noticed that in the *port* part, when disabling interrupts, previous BASEPRI value is stored in a variable and returned to user. Restoring them means write previous variable back.

But this is not the case in `schedule` assembly file that puts BASEPRI to `TX_PORT_BASEPRI` to lock irqs, and then forces BASEPRI back to `0` to unlock all.

Can this leads to an issue somehow?

CM7 port, considering TX_PORT_USE_BASEPRI is enabled (I removed ifdef code that is not used)

```c
__attribute__( ( always_inline ) ) static inline unsigned int __get_interrupt_posture(void)
{
unsigned int posture;
    __asm__ volatile ("MRS  %0, BASEPRI ": "=r" (posture));
    return(posture);
}

__attribute__( ( always_inline ) ) static inline void __set_basepri_value(unsigned int basepri_value)
{
    __asm__ volatile ("MSR  BASEPRI,%0 ": : "r" (basepri_value));
}

__attribute__( ( always_inline ) ) static inline void __restore_interrupt(unsigned int int_posture)
{
    __set_basepri_value(int_posture);
}

__attribute__( ( always_inline ) ) static inline unsigned int __disable_interrupts(void)
{
unsigned int int_posture;

    int_posture = __get_interrupt_posture();
    __set_basepri_value(TX_PORT_BASEPRI);
    return(int_posture);
}
```

But in the scheduler, disable IRQ:
https://github.com/azure-rtos/threadx/blob/54cda6ee9e68cd9701d759df13e3f6e41429b9bc/ports/cortex_m7/gnu/src/tx_thread_schedule.S#L136-L141

Re-enable IRQ
https://github.com/azure-rtos/threadx/blob/54cda6ee9e68cd9701d759df13e3f6e41429b9bc/ports/cortex_m7/gnu/src/tx_thread_schedule.S#L145-L150

MaJerle, 2022-07-11:
It is custom board with STM32H723 - USBX low-level port seems to be from MSFT straight. Part of X-CUBE-AZRTOS-H7 v2.1.0.
Enabling BASEPRI fails to work properly and I still don't know why. None of the interrupts have irq in register lower than `TX_PORT_BASEPRI`, meaning all are lower-or-equal to this one, aside millisecond timer, but that one doesn't call any OS-related functions (just increases custom time, not at all linked to OS).

goldscott, 2022-07-11:
I see that BASEPRI is set to 0 later in the scheduler (https://github.com/azure-rtos/threadx/blob/54cda6ee9e68cd9701d759df13e3f6e41429b9bc/ports/cortex_m7/gnu/src/tx_thread_schedule.S#L208-L213)

Can you please contact ST directly about this issue. We will also investigate.

MaJerle, 2022-07-11:
Question that remains - shall it be forced to zero or shall BASEPRI be stored first, then restored instead?

goldscott, 2022-07-11:
In the scheduler, BASEPRI is set to 0 before a thread is scheduled. This enables interrupts.

MaJerle, 2022-07-11:
Looking at FreeRTOs port for CM4F (CM7 is equal, aside core w/ bug) - there seems to be ISB or DSB after every MSR or MRS instruction.

MaJerle, 2022-07-12:
Found the problem - long night behind - BASEPRI is a masking register and therefore needs to have a proper mask set there, to determine level to block. I want to block all irqs with interrupt numerically higher or equal to `5`.

`TX_PORT_BASEPRI` is therefore a mask value, not a logical number.
If `4` priority bits are being used for interrupts, it means `16` levels, `0 - 15`. To properly use `TX_PORT_BASEPRI`, value has to be set to (assuming CMSIS is used) `BASEPRI = 5 << (8 - __NVIC_PRIO_BITS)`, 

For reference, FreeRTOS uses `2` values, from FreeRTOSConfig.h file:

```
/* Lowest numerical interrupt that can still use OS functions */
#define configLIBRARY_MAX_SYSCALL_INTERRUPT_PRIORITY	5

/* configMAX_SYSCALL_INTERRUPT_PRIORITY must not be set to zero - this is used for BASEPRI setup */
#define configMAX_SYSCALL_INTERRUPT_PRIORITY 	( configLIBRARY_MAX_SYSCALL_INTERRUPT_PRIORITY << (8 - configPRIO_BITS) )
```

At least an explanation could be added in your `tx_user` for instance, that we expect mask, not irq level value.

MaJerle, 2022-07-12:
Final conclusion, I use these settings in CMakeLists.

```
    "TX_INCLUDE_USER_DEFINE_FILE"
    "TX_PORT_USE_BASEPRI"                                   # Use BASEPRI feature for interrupt locking
    "TX_PORT_APP_MAX_IRQ_PRIO=5"                            # Minimum number for app interrupt priority using OS functions
                                                            # Interrupts with IRQ number lower than this (higher prio) are not allowed to call OS functions
    "TX_PORT_BASEPRI=(TX_PORT_APP_MAX_IRQ_PRIO << (4))"     # Eq: (level << (8 - __NVIC_PRIO_BITS))
                                                            # Convert user priority level to BASEPRI requested mask feature
                                                            # Check for MCU how many prio bits are available (__NVIC_PRIO_BITS in CMSIS-compliant software)
    "UX_INCLUDE_USER_DEFINE_FILE"
```

goldscott, 2022-10-28:
Hi @xiaocq2001 - can you please help with the above issue?

goldscott, 2022-11-09:
Hi @MaJerle - can you please contact ST about this issue.

closed 2022-12-20 by goldscott