eclipse-threadx/threadx #207: unit of trace event entry time stamp is too short for high cpu frequency

DBW19931021, 2022-08-08:
When useing the event tracex function of ThreadX, we find that the unit of time stamp of this function is 32 bits. In the case of high system frequency,  the recording time of the time stamp is very limited. May I ask whether ThreadX has a patch to modify relevant contents, such as changing 32-bit variables to 64 bit variables.

In file  tx_port.h:
TX_TRACE_TIME_MASK is 32 bits

 looking forward to your reply!

goldscott, 2022-08-08:
Hi @DBW19931021 - the time mask maximum is 32 bits, TraceX does not have support for 64-bit timers. 
What limitations are you running into? With a 1GHz clock, it will take 4 seconds for the timer to roll over, and TraceX handles timer rollovers.

DBW19931021, 2022-08-10:
Hi，@goldscott
We hope that we can observe the implementation of various tasks for a long time, for several minutes or even tens of minutes

goldscott, 2022-08-10:
TraceX knows when the timer wraps around, so you can do very long traces (assuming you have enough memory).

closed 2022-08-10 by goldscott

DBW19931021, 2022-08-10:
If I modify the kernel code to change the timestamp to 64 bits, what problems will happen？

goldscott, 2022-08-10:
The TraceX Windows application will not be able to read the file. It expects a 32-bit timestamp. Again: you do not need a 64-bit timer, TraceX knows how to handle the timer rollover.