eclipse-threadx/threadx #208: ThreadX with Modules support for TX_LOW_POWER support unclear

ascheurer, 2022-08-10:
tx_thread_schedule.S is different when using ThreadX modules compared to ThreadX. The threadx utilities lib supports example tx_low_power_enter and tx_low_power_exit routines. However, while declared in both tx_thread_schedule.S with ThreadX and ThreadX Modules, only ThreadX calls tx_low_power_enter/exit.

#ifdef TX_LOW_POWER
.global tx_low_power_enter
.global tx_low_power_exit
#endif
is defined in tx_thread_schedule.S for ports and ports_module however it is only called with BL using ports and not ports_module. Is TX_LOW_POWER supported with threadx modules? Unclear on why it would be defined w/ .global and never called in the tx_thread_schedule.S only for ports_modules (in particular gnu m4)

Does TX_LOW_POWER etc in https://github.com/azure-rtos/threadx/blob/master/utility/low_power/low_power.md apply to threadx modules?

How does low power work with threadx modules? 
Why declare it with the scheduler for threadx modules but not call it in tx_threadx_schedule.S ? 
Is there a plan to support?

goldscott, 2022-08-10:
Yes, the low power code will work with Modules. We simply haven't added the tx_low_power calls to the scheduler in Modules yet. I've added it to my list and we should have it available in the next release.

ascheurer, 2022-08-10:
Good news. Is there an approximate timeline for that next release?

goldscott, 2022-08-10:
The next release should be the end of October. If you intend to use Modules and low power in your application and need it sooner than that, please contact azure-rtos-support@microsoft.com

closed 2022-08-17 by goldscott