eclipse-threadx/threadx #209: Higher priority thread in ready state not become executing due to no SGI to inform the target core

meitlin, 2022-09-16:
Our system is 8-core SMP. We found a ready thread with highest priority (at that moment) not becoming executing.
Details are shown in the figure. (Task is same as thread in below description)
[image]

Task A running in core 4 was suspended from _tx_event_flag_set() and updated the _tx_thread_current_ptr[4] as null (M0). 
Then, core 4 went into _tx_thread_schedule() to schedule next Task (Task B). _tx_thread_current_ptr[4] was updated as Task B in _tx_thread_schedule() (M1). 
Task C running in core 5 set the event flag to resume Task A and all control variables were updated. However, Task C checked the _tx_thread_current_ptr[4] is null and then skipped sending SGI 0 to core 4 in _tx_thread_smp_core_interrupt() (M2).
As a result, Task A would not being scheduled even it is already in ready list with the highest priority and _tx_thread_execute_ptr[4] also updated as Task A. It seems there is a timing issue that cause core 5 not reading the lasted value of _tx_thread_current_ptr[4].

We do two experiments:
Experiment 1. Always send SGI 0 to target core without checking the _tx_thread_current_ptr[core x] is null or equal to _tx_thread_execute_ptr[core x].
Experiment 2. Add memory barrier (DMB ISH) after updating _tx_thread_current_ptr[core x] in _tx_thread_schedule()
It seems that both experiments could fix this issue.

However, we are not sure which solution is better or any other concern we might ignore?

Thanks.

meitlin, 2022-09-21:
Hi @goldscott , thanks for the quick reply. 
We use Cortex-A55-SMP/GNU, and the ThreadX version is 6.1.

carll00226, 2023-05-16:
hi, meitlin: 
    Does this problem caused by the store buffer between cpu and cache? if so, add barrier to flush the store buffer after write, do you also need to add barrier to clean the invalid queue before read int core5 ?

fdesbiens, 2026-09-09:
Hi @meitlin, apologies for the very long silence on this one. Your analysis was correct, and the good news is that the fix you proposed has been in ThreadX for a while now.

Your Experiment 2 is what was adopted. The barrier was added in commit `aea7b556a` ("Add DMB ISH barrier inst in ARMv8-A SMP scheduler") and shipped in the 6.2.1 release on 2023-03-08. In `_tx_thread_schedule` there is now a `DMB ISH` between the store to `_tx_thread_current_ptr[core]` and the reload of `_tx_thread_execute_ptr[core]`, which is exactly the window your diagram identifies. Since you were on a Cortex-A55 SMP GNU port, upgrading to 6.2.1 or later gives you the fix directly, and the current `dev` branch also carries it.

For the record, the reason Experiment 2 is preferred over Experiment 1 is the one @goldscott gave: unconditionally sending SGI 0 to every target core works, but it generates an inter-core interrupt on every scheduling decision even when the target core is already running the right thread, and on an 8-core system that overhead is significant. The barrier costs a few cycles on the scheduling path only.

@carll00226, on your question: on Arm this is not really a store-buffer flush in the x86 sense. `DMB ISH` is an ordering instruction, not a cache maintenance one. Coherency between cores is handled by the hardware, so no explicit invalidate is required on the reading side, and there is no need for a separate barrier before the read in `_tx_thread_smp_core_interrupt` to make the data visible. What the barrier guarantees is that the write to `_tx_thread_current_ptr[core]` is ordered before the subsequent read of `_tx_thread_execute_ptr[core]` on the same core, which is what makes the recheck-and-back-out logic in the scheduler reliable.

While confirming this I also found that the ARMv7-A SMP ports (Cortex-A5, Cortex-A7 and Cortex-A9, both AC5 and GNU) implement the same recheck protocol but never received the matching barrier when the ARMv8-A ports did. I have opened #704 to correct that.

I am closing this issue since the reported problem is fixed. Please do reopen it or file a new one if you still see the symptom on a recent release.

closed 2026-09-09 by fdesbiens