eclipse-threadx/threadx #213: RISCV GNU port support

lingjian0514, 2022-10-11:
HI: 
   when will you add support for riscv gun port and gnu smp_port?

goldscott, 2022-10-11:
We should have a RISC-V GNU port at the end of October/beginning of November!

goldscott, 2022-10-28:
Hi @lingjian0514 - we are still working on the GNU port. We are targeting both 32-bit and 64-bit architectures, as well as the various options (e.g., none/single/double floating point), so testing is taking a bit longer.

carll00226, 2022-11-01:
Hi scott:
    We are about to port threadx to a riscv smp system, and we have simulation for a riscv platform. Do you have an experimental version of GNU SMP threadx which we can refer to , and we can also do some test for it?

goldscott, 2022-11-01:
Hi @carll00226 - we do not have a RISC-V SMP port yet. I will let you know when one is available!

carll00226, 2022-11-18:
Hi，scott:
      I have finished a threadx port to a riscv smp system, but there are some bugs when testing.  
      I create 2 tasks, task A send a message queue and then call tx_thread_sleep(5) in a while loop, and task B just receive the message. after the tx_thread_sleep, both task A and task B will sleep, and call tx_thread_system_suspend->tx_thread_system_return to schedule loop, and will clear tx_thread_current_ptr to 0 in tx_thread_system_return.  After 5 ticks, it will call tx_timer_interrupt->tx_timer_expiration_process->tx_thread_system_resume->tx_thread_system_return, and in tx_thread_system_return will get tx_thread_current_ptr  and set sp to it , but now tx_thread_current_ptr  is NULL.  I only find the tx_thread_current_ptr will be set in tx_thread_schedule, and don't know what's wrong with my process.  Would you please give me some advice.
      Thanks so much!

carll00226, 2022-11-19:
I have fixed the bug above.

goldscott, 2022-12-20:
Hi @carll00226 - have you pushed your port anywhere? I'd love to take a look.

wanglehui, 2023-01-11:
Hi scott:
     When will you release the RISC-V GNU port?

wanglehui, 2023-03-09:
Why Threadx port for RISC-V part is so sloppy？
(1) RV64  port is bad for __riscv_float_abi_single
(2) Both rv32 and rv64 dot have normal trap handle function  (what is the timer1_plic_IRQHandler or __minterrupt_000007???)

why not supply unified port for both RV32/RV64 and FPU-Single/Double/None ?

goldscott, 2023-04-12:
@wanglehui - I am no longer on the Azure RTOS team, so I was not able to complete the 64-bit port. It seems it was released prematurely. There are still some bugs in the floating point that Microchip is helping us fix. The 64-bit port should also work with 32-bit architectures.

fdesbiens, 2025-02-27:
Closing this issue given to the lack of activity. Please open a new issue if you are interested in a conversation on RISC-V support. 

Improving RISC-V support is an important goal for the project; however, our current committers will work on what their customers and partners request. 

We will welcome new contributors with proven RISC-V expertise who are willing to help us improve our RISC-V ports.

closed 2025-02-27 by fdesbiens