eclipse-threadx/threadx #215: Missing a curly bracket?

mikisama, 2022-10-24:
Hi,

The comments in the assembly code seem to be missing a curly bracket.
Not only the risc-v32 iar port.

```diff
diff --git a/ports/risc-v32/iar/src/tx_thread_context_restore.s b/ports/risc-v32/iar/src/tx_thread_context_restore.s
index 8a23b6e..3d497f2 100644
--- a/ports/risc-v32/iar/src/tx_thread_context_restore.s
+++ b/ports/risc-v32/iar/src/tx_thread_context_restore.s
@@ -180,7 +180,7 @@ _tx_thread_context_restore:
     /* }  */
 _tx_thread_not_nested_restore:
     /* Determine if a thread was interrupted and no preemption is required.  */
-    /* else if (((_tx_thread_current_ptr) && (_tx_thread_current_ptr == _tx_thread_execute_ptr) 
+    /* else if (((_tx_thread_current_ptr) && (_tx_thread_current_ptr == _tx_thread_execute_ptr))
                || (_tx_thread_preempt_disable))
```

[image]

goldscott, 2022-10-24:
Hi @mikisama - thanks for finding this! I'll close this ticket once we have a release that fixes these comments.

TiejunMS, 2023-03-14:
Fixed in 6.2.1 release. Closing.

closed 2023-03-14 by TiejunMS