eclipse-threadx/threadx #216: [HELP WANTED] Linker script in case of loading from memory.

zq649, 2022-11-07:
Arch: cortex_m7
Toolchain: GNU

Hi,
I have general question regarding linker script. If the module is fetched in a dynamic memory from a file system, how can I identify the starting address for FLASH or RAM to write in my linker script?

goldscott, 2022-11-07:
You can use any address you want for the linker script. For example, [here](https://github.com/azure-rtos/threadx/blob/master/ports_module/cortex_m7/gnu/example_build/sample_threadx_module.ld) we use 0x00030000.

zq649, 2022-11-07:
@goldscott Thanks for the response. About the regions, I guess I can only use one or both FLASH and RAM would be required?

goldscott, 2022-11-08:
@zq649 you can execute a module from flash or RAM. It's up to you to determine where to place it. If you are going to read a module from a file and flash it, you would need to implement the code necessary to write to your chip's flash. Then you should call txm_module_manager_in_place_load. If you want to execute from RAM, you can simply call txm_module_manager_file_load and it will read the module from the file and place the module in RAM

goldscott, 2022-12-14:
@qasim-ublox ah, you need to add ".section .preamble" to the preamble assembly file like is done here:
https://github.com/azure-rtos/threadx/blob/master/ports_module/cortex_m0%2B/gnu/example_build/sample_threadx_module/txm_module_preamble.S#L4

qasim-ublox, 2022-12-14:
@goldscott thanks. Any specific reason that this line is not added in preamble file for cortexm7?

goldscott, 2022-12-14:
@qasim-ublox I think it was just forgotten in the M3/M4/M7 ports. I'll make a note to add it for our next release.

qasim-ublox, 2022-12-14:
@goldscott this [line](https://github.com/azure-rtos/threadx/blob/master/ports_module/cortex_m0+/gnu/example_build/sample_threadx_module/sample_threadx_module.ld#L102) is also not added for M7.

goldscott, 2022-12-14:
0x80 is the length of the preamble in bytes.

wangwen-4220, 2023-06-27:
@qasim-ublox, we have a module(cm7,gcc) project here
https://github.com/wangwen-4220/stm32f746-disco-module/blob/main/stm32cubeide/sample_threadx_module/cm7/sample_threadx_module.ld

we have done some fix about the sample_threadx_module.ld and txm_module_preamble.S.

please let us know if it is helpful and we can close the issue.

fdesbiens, 2026-09-11:
Hi @zq649, @qasim-ublox, @wangwen-4220 — apologies for the long silence on this one. Everything raised in this thread has since been addressed, so I am closing it.

On the original question: as @goldscott explained, the module's link address is entirely your choice. If you read a module from a file system and execute it in place from flash, you write it to flash yourself and then call `txm_module_manager_in_place_load()`. If you want to execute from RAM, `txm_module_manager_file_load()` reads the module and places it in RAM for you. You need a FLASH region in the linker script only for the in-place case; a RAM-resident module needs only the RAM region.

On the missing `.section .preamble` in the M3/M4/M7 GNU preambles that @qasim-ublox reported: that was fixed in release 6.2.1. `ports_module/cortex_m{3,4,7}/gnu/example_build/txm_module_preamble.S` now declares `.section .preamble, "ax"`, and the matching `sample_threadx_module.ld` places `*(.preamble .preamble.*)` first, at `__FLASH_segment_start__`. That is the GNU equivalent of the ARMCC `--first` option and guarantees the preamble lands at offset 0 of the module image.

On the `+ 0x80` in the Cortex-M0+ script: that is the fixed 128-byte length of the module preamble. The M0+ script places the preamble inside `.text` and therefore has to compute `__preamble_end__` from the known preamble size rather than from `SIZEOF(.preamble)`. The M3/M4/M7 scripts give the preamble its own output section at the start of flash, so they can use `SIZEOF(.preamble)` directly and do not need the constant. Both arrangements are valid; the M3/M4/M7 one is the clearer of the two.

@wangwen-4220, thank you for sharing your STM32F746-Disco module project — it is a useful reference for anyone bringing up a Cortex-M7 GNU module build.

Please open a new issue if you hit anything further.

closed 2026-09-11 by fdesbiens