eclipse-threadx/threadx #217: Cortex-M55 ac6: Compilation error due to arm_compat.h include

jite, 2022-11-18:
I'm trying to compile latest master of threadx (as static library) for `cortex-m55,` using `armclang` version 6+.

I'm running into compilation issues where the include of `arm_compat.h` (`ports/cortex_m55/ac6/inc/tx_port.h`) causes conflicts with CMSIS definitions for `__enable_irq` and `__disable_irq`.

Specifically in the `cortex-m55` port I can not find that the use of `arm_compat.h` is necessary, removing the include solves the issue for me.

goldscott, 2022-11-18:
Thanks for the report. We'll remove arm_compat.h from the include. This will be done in the next ThreadX release.

goldscott, 2023-01-24:
This has been fixed https://github.com/azure-rtos/threadx/commit/37f6d0b39c970b297c263cbb8388b9060b92e5c9

closed 2023-01-24 by goldscott