eclipse-threadx/threadx #218: Unaligned writes are not allowed with gcc ubsan (-fsanitize=undefined)

owbear, 2022-11-24:
https://github.com/azure-rtos/threadx/blob/b42c5acd8b9b1eb156c25e0b18c82beecfc033a1/ports/linux/gnu/src/tx_thread_stack_build.c#L131

The write here gives me an error regarding unaligned writes. `memset()` works

goldscott, 2022-12-01:
You can ignore this. The stack is aligned previously - https://github.com/azure-rtos/threadx/blob/b42c5acd8b9b1eb156c25e0b18c82beecfc033a1/common/src/tx_thread_create.c#L132

owbear, 2022-12-01:
A bit more context here: in the linux/gnu port `tx_thread_stack_ptr` is always unaligned as it is `tx_thread_stack_end-8`:

```    /* Setup a fake thread stack pointer.   */
    thread_ptr -> tx_thread_stack_ptr =  (VOID *) (((CHAR *) thread_ptr -> tx_thread_stack_end) - 8);

    /* Clear the first word of the stack.  */
    *(((ULONG *) thread_ptr -> tx_thread_stack_ptr) - 1) =  0;```

owbear, 2022-12-16:
Hi Scott! `TX_ENABLE_STACK_CHECKING` fails in a different way with ubsan enabled using ports/linux/gnu:

    Starting program: arch/native/radiohub 
    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
    [New Thread 0xf4affb40 (LWP 37588)]
    [New Thread 0xf3effb40 (LWP 37589)]
    [New Thread 0xf36feb40 (LWP 37590)]
    /home/gotber/src/radiohub/code-review/env/threadx/common/src/tx_thread_system_resume.c:108:5: runtime error: load of misaligned address 0x56815217 for type 'ULONG', which requires 4 byte alignment
    0x56815217: note: pointer points here
     ef ef ef ef 00  00 00 00 ef ef ef ef ef  ef ef ef ef ef ef ef ef  d8 60 81 56 ee ee ff ff  00 00 00
                 ^ 
    [Thread 0xf36feb40 (LWP 37590) exited]
    [Thread 0xf4affb40 (LWP 37588) exited]
    [Thread 0xf7012440 (LWP 37584) exited]
    [Inferior 1 (process 37584) exited with code 01]
    (gdb) l env/threadx/common/src/tx_thread_system_resume.c:108
    103	
    104	
    105	#ifdef TX_ENABLE_STACK_CHECKING
    106	
    107	    /* Check this thread's stack.  */
    108	    TX_THREAD_STACK_CHECK(thread_ptr)
    109	#endif
    110	
    111	    /* Lockout interrupts while the thread is being resumed.  */
    112	    TX_DISABLE

I can look into why; it wasn't immediately obvious.

owbear, 2022-12-16:
That misalignment could be resolve with `memcmp()`

        static ULONG tx_stack_fill = TX_STACK_FILL;
        if (0 != memcmp(&tx_stack_fill, (thread_ptr)->tx_thread_stack_highest_ptr - 1, sizeof tx_stack_fill))

fdesbiens, 2026-09-09:
Hi @owbear, apologies for the very long silence on this one. You were right on both counts, and the underlying cause is the same for both reports.

`tx_thread_stack_end` addresses the last byte of the thread's stack area, so its value is one less than an aligned address. The simulation ports derive their fake initial stack pointer from it with a plain `- 8` and never re-align the result, which means that pointer is misaligned by construction regardless of how well aligned the stack the application passed in was. That is the misaligned store you first hit in `tx_thread_stack_build.c`.

The second failure follows from the same pointer: `_tx_thread_create()` copies it into `tx_thread_stack_highest_ptr`, and `TX_THREAD_STACK_CHECK` dereferences that as a `ULONG` on every suspend and resume, which is the `tx_thread_system_resume.c` diagnostic you reported with `TX_ENABLE_STACK_CHECKING` enabled.

I reproduced both messages exactly as you described them by building `ports/linux/gnu` with `-fsanitize=undefined`. The fix rounds the fake stack pointer down to a `ULONG` boundary, which keeps it inside the stack area and makes both accesses aligned, so no `memset`/`memcmp` workaround is needed and the checking logic keeps scanning on natural boundaries. The same change was applied to the Win32, Win64 and SMP Linux ports, which all share the code.

The fix is in PR #705. Thank you for the clear report and for following up with the second case.

closed 2026-09-09 by fdesbiens