eclipse-threadx/threadx #219: Can TX_DISABLE/TX_RESTORE be called in interrupt handler in smp system?

carll00226, 2022-11-28:
Hi, scott:
    I'm running threadx in a two-core smp sytem of a arm 32-bit platform,  and get dead lock  while two cores are all in already_in_list1 of tx_thread_smp_protect.S.  One of the core is in interrupt handler and called TX_DISABLE while the IRQs are disabled. I don't know if the tx_thread_smp_wait_list should be changed relied on the SGI, but as the IRQs are disabled,  the core can't generate SGI to modify the  tx_thread_smp_wait_list.
   I want to know if TX_DISABLE/TX_RESTORE can be called in isr while IRQs are disabled?  Because if the core cannot get locked in  tx_thread_smp_protect, it will execute MSR  CPSR_c, r0 to restore CPSR to generate SGI.  And what about other APIs such as tx_semaphore_get which will call TX_DISABLE in it's context? 
   Thanks.

fdesbiens, 2026-09-09:
Hi @carll00226, and my apologies that this sat for so long after Scott's reply.

To answer your direct question first: yes, `TX_DISABLE`/`TX_RESTORE` are safe to call from an ISR, including one entered with interrupts already masked. They save and restore the interrupt posture rather than forcing it, so on an already-masked core they are effectively no-ops. The same is true of the services that use them internally, such as `tx_semaphore_get`. That part was never the problem.

The problem was the FIFO wait list in the ARMv7-A `_tx_thread_smp_protect`, and your diagnosis was correct. Two parts of that scheme need the waiting core to be interruptible: `_tx_thread_smp_unprotect` will not release the protection while the releasing core's own wait count is non-zero, and a queued core is only taken back out of the list by `_tx_thread_context_restore` when it gets preempted. A core spinning in `_already_in_list1` inside an ISR has interrupts masked and the `MSR CPSR_c, r0` in that loop restores that same masked posture, so it can never be preempted and never leaves the list. That is the deadlock you saw.

As Scott said, this was fixed for ARMv8-A in 6.1.11 by removing the wait list altogether in favour of a simple exclusive-access spinlock. The matching ARMv7-A change was announced then but unfortunately never landed, which is why you are still hitting it.

PR #707 applies that same removal to the Cortex-A5, A7 and A9 SMP ports. The new `_tx_thread_smp_protect` is an LDREX/STREX spinlock that genuinely restores interrupts between attempts, matching what the Cortex-R8 SMP port already does. I have verified it by assembly and disassembly, but I do not have ARMv7-A SMP hardware here, so if you are still working with this platform a test against your workload would be extremely valuable.

Thank you for the clear and well-reasoned report.

fdesbiens, 2026-09-09:
PR #707 has been merged, so the FIFO wait list is gone from the Cortex-A5, A7 and A9 SMP ports and this deadlock can no longer occur. Closing the issue given its age, but please reopen or file a new one if you still see anything like it on ARMv7-A SMP hardware.

closed 2026-09-09 by fdesbiens