eclipse-threadx/threadx #222: [General Question] Regarding -fpic flag

qasim-ublox, 2023-01-09:
Toolchain : GCC-ARM

Hi,
I've seen that `-fpic` flag is being used in batch script files for GNU port, but I don't see the flag in toolchain file?

goldscott, 2023-01-09:
What batch script files are you seeing that flag, and what toolchain files are you not seeing it?

qasim-ublox, 2023-01-09:
For example, [here](https://github.com/azure-rtos/threadx/blob/4e62226eeaf870827facbeef40bd1767db5cf9f0/ports_module/cortex_m7/gnu/example_build/build_threadx_module_library.bat) it is being used.
But [here](https://github.com/azure-rtos/threadx/blob/master/cmake/arm-none-eabi.cmake) in the toolchain file it is not used.

goldscott, 2023-01-11:
I'll close this ticket now. If you have further questions, feel free to re-open.

closed 2023-01-11 by goldscott