eclipse-threadx/threadx #223: New licensed hardware

FernanGallego, 2023-01-16:
Is there any plan to add more hardware in the LICENSED-HARDWARE list? 
It was said to be more coming soon, but there has not been an update sin April 2021. I'm planning o a new HW design and that would be a great information to know.

Kind regards,

goldscott, 2023-01-19:
We don't have any new partners/hardware yet. What MCUs are you considering?

FernanGallego, 2023-01-23:
Thanks for your response Scott. We are testing some ARM from NXP. We are
actually working with the IMX8MPlus eval kit. But we could use some other
ARM as long as it has PCIExpress ports.

Kind regards,

*Fernando Gallego Vilaltella*

Operations and Innovation Manager

M  (+34) 654 93 41 54

T  (+34) 932 98 86 50

E  fernan.gallego @ lanaccess. com

LANACCESS <https://www.lanaccess.com/> | + segura + inteligente + rentable

goldscott, 2023-01-24:
I suggest you reach out to NXP to see if they are considering adding the iMX8 to the licensed hardware list. Offhand, I don't know what currently licensed hardware has PCIE ports.

closed 2023-01-30 by goldscott