eclipse-threadx/threadx #226: tx_thread_sleep

netjoon, 2023-01-26:
**Describe the bug**
tx_thread_sleep is not accurate when I set below system_clock and systick_cycles.

SYSTEM_CLOCK      =   491520000
SYSTICK_CYCLES    =   ((SYSTEM_CLOCK / 1000000) -1)

If I defined above, 1 systick is 1 us
When I call tx_thread_sleep(10) I'm expecting my system to wake up 10 systick later.
I added tx_time_get() before and after sleep and check the tick, it varies from 17to 35.
Is this expected behavior?
If it is, how can we have exact us(microsecond)-based sleep time?

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? 5G Modem Chip
- Which version of Azure RTOS? 6.1.12 cortex-m3 version
- What toolchain and environment? arm-none-eabi-gnu
- What have you tried to diagnose or workaround this issue? 

**To Reproduce**
Steps to reproduce the behavior:
1. Build my project in IAR Workbench.
Set below values in tx_initialize_low_level.S
SYSTEM_CLOCK      =   491520000
SYSTICK_CYCLES    =   ((SYSTEM_CLOCK / 1000000) -1)
1. See error
ULONG old, new;
old = tx_time_get();
tx_thread_sleep(10);
new = tx_time_get();
printf("sleep time: %ld\n", new - old);

**Expected behavior**
I'm expecting print value around 10 but it is vary between 16 to 33.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Can not use this RTOS.

**Logs and console output**
If applicable, add console logs or other types of debug information like Wireshark capture as `.zip` file.

**Additional context**
Add any other context about the problem here.

netjoon, 2023-01-28:
This is not a tx_thread_sleep issue. I found root cause.

closed 2023-01-28 by netjoon

goldscott, 2023-01-30:
Hi @netjoon - what was the root cause?

netjoon, 2023-01-30:
I've set two thread's priority same. One thread sleep and the other thread is running. When 1st thread try to wake up, 2nd thread which has same priority with 1st thread is running. No preemption happened. That's why this issue happened.